travisgoodspeed/md380tools #270: proposal for firmware defines

sijskes, 2016-09-08:
To conditional include code i propose:

#define FW_D13_020  1

to be passed when compiling for d13.020 fw. etc....

travisgoodspeed, 2016-09-08:
Welcome to the team, neighbor!

Approved.  Close this issue when each version has its definition.

sijskes, 2016-09-08:
Thanks for inviting me.

Created a wiki page with the defines 'Coding Notes'. 

Will close when it is in makefile

aeickho, 2016-09-08:
I prefer "my" style.. it's the same as the folder structure and there is only one change in the makefile necessary.

See on makefile ....

aeickho, 2016-09-09:
See also my comments #272 ;-)

roelandjansen, 2016-09-09:
I believe that sijske's idea is a good thing.

aeickho, 2016-09-09:
see:
https://github.com/travisgoodspeed/md380tools/blob/master/applet/Makefile#L38
and:
https://github.com/travisgoodspeed/md380tools/blob/master/applet/src/beep_process.c

sijskes, 2016-09-09:
roger. did not see that as firmware specific. more hardware specific. when you mention MD380 in a define, this can happen.

aeickho, 2016-09-15:
@sijskes  Is your define to "1" important for your concept?

my idea ...
CFLAGS += -D "MD380_$(subst s,S,$(subst d,D,$(subst .,_,$(SRCVERSION))))"
then the new "based on fw define" is:

**MD380_D13_020**

sijskes, 2016-10-06:
implemented proposal

closed 2016-10-06 by sijskes