travisgoodspeed/md380tools #287: Have automated builds using github releases

hsmade, 2016-09-12:
It would be nice if we could have automated builds for the firmware that get uploaded to the github releases page.

https://docs.travis-ci.com/user/deployment/releases

hamradioPRC, 2016-09-13:
https://pd0zry.nl/md380-fw/

hsmade, 2016-09-13:
Rudy builds every hour, if there are new commits. What I am proposing different because:
 * it will build directly after the tag/commit (to be decided)
 * it will deliver the artifact (the bin file(s)) to the github releases page, so it's all on the same place.

travisgoodspeed, 2016-09-14:
Can github releases be precompiled, or are they only the source code of tagged releases?

travisgoodspeed, 2016-09-14:
Issue #290 was caused by PD0ZRY's sorting of firmware versions, which put S13.020 ahead of the others even though it won't boot on most radios.

We really need a properly organized download page to prevent future confusion.

hsmade, 2016-09-14:
here is an example: https://github.com/roelandjansen/md380tools/releases

Yes, releases can contain binaries, that is their purpose actually.
What I do on my own pipelines is tag my master with a version number (https://semver.org) and publish that.

hsmade, 2016-09-14:
"We really need a properly organized download page to prevent future confusion."

This is exactly the reason why I proposed this, actually :)

travisgoodspeed, 2016-09-14:
Okay, we'll enable these builds and start scheduling toward a release candidate.

Cheers,
6A/KK4VCZ

roelandjansen, 2016-09-14:
We have a working scenario where jenkins builds direct after I push and the release is then.on my page

hamradioPRC, 2016-10-13:
Nothing new as of yet?

I assume fw build 3FFA38F is last one.

roelandjansen, 2016-10-14:
the last builds are integrated at 

https://github.com/roelandjansen/md380tools/releases

and the last stable at this moment is

https://github.com/roelandjansen/md380tools/releases/tag/e762fc9

There is internalwork going on at this moment and not published yet.

CoMPaTech, 2016-10-14:
For more information on what we are doing check https://github.com/travisgoodspeed/md380tools/blob/master/ContinuousIntegration.md

hamradioPRC, 2016-10-16:
What is being done with travisgoodspeed/md380tools/releases?

No tags/builds been setup yet. Only Roelandjansens commits are built to master on his github with hourly pre/latest releases

roelandjansen, 2016-10-16:
Correct. The ci/cd onlt runs on that one