travisgoodspeed/md380tools #290: Latest Firmware makes my MD380 does not start.

PD1LOI, 2016-09-14:
Latest Firmware makes my MD380 does not start. just a blank white screen is what you see.
Firmware 973K big works.
Firmware 1000 K big does not work.

travisgoodspeed, 2016-09-14:
We really, *really* need more information with bug reports.  Otherwise we can't reproduce your problem, and that's frustrating for both of us.

I've just verified that both 'make flash' and 'make flash_d02.032` correctly flash functional firmware into a non-GPS MD380.  It is a known issue, tracked by #181, that `make flash_s13.020` only boots on devices with GPS hardware.

We don't have a firmware that is 1000 kilobytes in size.  Could you please provide us with more details of your versions and how you are trying to compile?

73 from Varaždin,
6A/KK4VCZ

PD1LOI, 2016-09-14:
make -C lib clean make[1]: Entering directory
'/home/zarya/md380tools/applet/lib' rm -f misc.o stm32f4xx_dma.o
stm32f4xx_rcc.o stm32f4xx_adc.o stm32f4xx_exti.o stm32f4xx_rng.o
stm32f4xx_can.o stm32f4xx_flash.o stm32f4xx_rtc.o stm32f4xx_crc.o
stm32f4xx_fsmc.o stm32f4xx_sdio.o stm32f4xx_cryp_aes.o stm32f4xx_gpio.o
stm32f4xx_spi.o stm32f4xx_cryp.o stm32f4xx_hash.o stm32f4xx_syscfg.o
stm32f4xx_cryp_des.o stm32f4xx_hash_md5.o stm32f4xx_tim.o
stm32f4xx_cryp_tdes.o stm32f4xx_hash_sha1.o stm32f4xx_usart.o
stm32f4xx_dac.o stm32f4xx_i2c.o stm32f4xx_wwdg.o stm32f4xx_dbgmcu.o
stm32f4xx_iwdg.o stm32f4xx_dcmi.o stm32f4xx_pwr.o libstm32f4.a make[1]:
Leaving directory '/home/zarya/md380tools/applet/lib' rm -f *.img *.lst
*.bin *.elf *.hex *.o *~ *.sym src/*~ src/version.h echo "#define
VERSIONDATE L\"`date "+%Y-%m-%d"`\"" >src/version.h make -C lib make[1]:
Entering directory '/home/zarya/md380tools/applet/lib' arm-none-eabi-gcc -g
-O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork
-mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc
-Iinc/core -Iinc/peripherals -fshort-wchar -c -o misc.o
src/peripherals/misc.c arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian
-mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard
-mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core
-Iinc/peripherals -fshort-wchar -c -o stm32f4xx_dma.o
src/peripherals/stm32f4xx_dma.c arm-none-eabi-gcc -g -O2 -Wall
-mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard
-mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core
-Iinc/peripherals -fshort-wchar -c -o stm32f4xx_rcc.o
src/peripherals/stm32f4xx_rcc.c arm-none-eabi-gcc -g -O2 -Wall
-mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard
-mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core
-Iinc/peripherals -fshort-wchar -c -o stm32f4xx_adc.o
src/peripherals/stm32f4xx_adc.c arm-none-eabi-gcc -g -O2 -Wall
-mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard
-mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core
-Iinc/peripherals -fshort-wchar -c -o stm32f4xx_exti.o
src/peripherals/stm32f4xx_exti.c arm-none-eabi-gcc -g -O2 -Wall
-mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard
-mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core
-Iinc/peripherals -fshort-wchar -c -o stm32f4xx_rng.o
src/peripherals/stm32f4xx_rng.c arm-none-eabi-gcc -g -O2 -Wall
-mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard
-mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core
-Iinc/peripherals -fshort-wchar -c -o stm32f4xx_can.o
src/peripherals/stm32f4xx_can.c arm-none-eabi-gcc -g -O2 -Wall
-mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard
-mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core
-Iinc/peripherals -fshort-wchar -c -o stm32f4xx_flash.o
src/peripherals/stm32f4xx_flash.c arm-none-eabi-gcc -g -O2 -Wall
-mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard
-mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core
-Iinc/peripherals -fshort-wchar -c -o stm32f4xx_rtc.o
src/peripherals/stm32f4xx_rtc.c arm-none-eabi-gcc -g -O2 -Wall
-mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard
-mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core
-Iinc/peripherals -fshort-wchar -c -o stm32f4xx_crc.o
src/peripherals/stm32f4xx_crc.c arm-none-eabi-gcc -g -O2 -Wall
-mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard
-mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core
-Iinc/peripherals -fshort-wchar -c -o stm32f4xx_fsmc.o
src/peripherals/stm32f4xx_fsmc.c arm-none-eabi-gcc -g -O2 -Wall
-mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard
-mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core
-Iinc/peripherals -fshort-wchar -c -o stm32f4xx_sdio.o
src/peripherals/stm32f4xx_sdio.c arm-none-eabi-gcc -g -O2 -Wall
-mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard
-mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core
-Iinc/peripherals -fshort-wchar -c -o stm32f4xx_cryp_aes.o
src/peripherals/stm32f4xx_cryp_aes.c arm-none-eabi-gcc -g -O2 -Wall
-mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard
-mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core
-Iinc/peripherals -fshort-wchar -c -o stm32f4xx_gpio.o
src/peripherals/stm32f4xx_gpio.c arm-none-eabi-gcc -g -O2 -Wall
[... 217 lines cut ...]
ambe.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of
wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9/../../../arm-none-eabi/bin/ld: warning:
dmr.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of
wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9/../../../arm-none-eabi/bin/ld: warning:
lib/libstm32f4.a(stm32f4xx_gpio.o) uses 2-byte wchar_t yet the output is to
use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.9/../../../arm-none-eabi/bin/ld: warning:
lib/libstm32f4.a(stm32f4xx_rcc.o) uses 2-byte wchar_t yet the output is to
use 4-byte wchar_t; use of wchar_t values across objects may fail
arm-none-eabi-objcopy -O binary main.elf main.img arm-none-eabi-objdump -t
main.elf >main.img.sym cd ../patches/s13.020 && make all make[1]: Entering
directory '/home/zarya/md380tools/patches/s13.020' make[1]: Nothing to be
done for 'all'. make[1]: Leaving directory
[... cut ...]
all make[1]: Entering directory '/home/zarya/md380tools/patches/s13.020'
make[1]: Nothing to be done for 'all'. make[1]: Leaving directory
'/home/zarya/md380tools/patches/s13.020' cp ../patches/s13.020/patched.img
./experiment.img python2 merge_s13.020.py experiment.img main.img 0x809c000
Merging an applet. Loading symbols from main.img.sym Merging main.img into
experiment.img at 0809c000 ../md380-fw --wrap experiment.img experiment.bin
DEBUG: reading "experiment.img" INFO: base address 0x800c000 INFO: length
0xf4000 DEBUG: writing "experiment.bin"

PD1LOI, 2016-09-14:
Look here https://pd0zry.nl/md380-fw/

travisgoodspeed, 2016-09-14:
Use the D13.020 firmware, not the S13.020 firmware.  S13.020 is only for GPS, and it does not work on radios without GPS.

`firmware-20160913-220001-a4523e2a.bin` will work fine.

`firmware-S13.020-20160913-220001-a4523e2a.bin` will not work.

Closing issue.

closed 2016-09-14 by travisgoodspeed

PD1LOI, 2016-09-14:
Yes i see the fault on my site.
Thanks