travisgoodspeed/md380tools #292: update readme.mk ... difference about D and S Version

aeickho, 2016-09-14:
to reduce misunderstandings :) 

more documentation is necessary 

close this item when we have an list of radio types and the corresponding md380tools fw

travisgoodspeed, 2016-09-14:
Alternately, as a better fix, close this issue if we can get S013.020 booting on non-GPS hardware.  (Or D13.020 booting on GPS hardware.)

#181

aeickho, 2016-09-14:
I have ordered an gps radio yet...  and then i start to work on it ;)

bendikz, 2016-09-14:
Ideally, I think that the D-version should boot on both GPS and non-GPS as I expect that the new GPS functionality would conflict on non-GPS radios. Otherwise we just got to be clear about that D is for non-GPS and S is for GPS. 

Could a wizard type of application be desirable? Like a script asking what radio you got and whether you just want the patched firmware, to flash the firmware, update Contact IDs etc?

travisgoodspeed, 2016-09-14:
I'm sure we'll eventually patch D13.020 to work on both, but for now,, 2.032 is the only version that boots on both GPS and non-GPS devices.

Wizards don't help when people get their firmware from many different sources.

I've begun a page in the wiki for downloading prebuilt firmware.  Close this issue when it is complete.
https://github.com/travisgoodspeed/md380tools/wiki/Download-Firmware

travisgoodspeed, 2016-09-14:
Updated the wiki page with links to PD0ZRY's build page.

Created Issue #294 to track better Windows distributables.

travisgoodspeed, 2016-09-14:
#294 is closed, and you can now run 'make dist' to produce a single ZIP with all three firmware builds and a Windows installer.

To minimize confusion, the firmware images are called "NoGPS", "GPS", and "OLD".

aeickho, 2016-12-12:
I hear no problems in this regard

closed 2016-12-12 by aeickho