travisgoodspeed/md380tools #295: usb stability during dmesgtail

sijskes, 2016-09-14:
when pulling the 'side-plug' from the md380 during dmesgtail, there is a high prob of hanging the md380.

bendikz, 2016-09-14:
Could you please elaborate a bit, what radio, firmware and PC did you experience this with? I have also experienced that the same applies to the other usb interfacing tools if you are to quick pulling the cable out. This was a MD390 GPS w S013.020 firmware and 64bit Ubuntu 16.04.

sijskes, 2016-09-14:
md380. ubuntu install conform recipe. 64bits. 16.04. d13.020. high traffic dmesg printf local patch.

travisgoodspeed, 2016-09-14:
Could be a complicated threading thing, but it might also be that `dmesg_tx_buf` is reused by a lot of different functions.  We need to clean up the linking description to be able to conveniently place buffers in both SRAM and TCRAM.

(TCRAM at 0x10000000 cannot be used for DMA, but we have more if it than the SRAM at 0x20000000.)

sijskes, 2016-09-14:
i have the impression that pulling the plug glitches the PTT. anybody confirming?

travisgoodspeed, 2016-09-14:
Oh yeah, that'd do it.  I've had some luck with a ferrite choke around the USB cable, but realistically, I don't think the hardware is stable enough to transmit with the default cable.

Close this issue if we can confirm it's not a software problem.

aeickho, 2016-09-25:
also  ./md380-tool spiflashdump makes problems

aeickho, 2016-09-25:
next try... remove all none usb hooks

aeickho, 2016-09-25:
(remove all none usb hooks) same problem

aeickho, 2016-09-25:
perhaps there is a transfer errors from d02.32 to [DS]13.020 :(

sijskes, 2016-09-25:
@aeickho no, the dmesgtail on d13 works. but slows down gui, considerable.

ispkorte, 2016-09-25:
on my RT8 S13 woks flash whitout error or warning

aeickho, 2016-09-25:
> sijskes > @aeickho no, the dmesgtail on d13 works. but slows down gui, considerable.
 yes this is true

but `./md380-tool spiflashdump` makes problems  also with d13.020

roelandjansen, 2016-09-25:
Alex, as we speak I am dumping my md380 into a file and the time / menu works OK.
This is under vmware workstation, vm is debian 8.x and the virtualization host is opensuse. 

Could you tell me if we do the same thing?

aeickho, 2016-09-25:
@roelandjansen i use a native linux system.

dump the 16Mbyte spi flash to the pc:

`./md380-tool spiflashdump file.bin`

my idea ... your system is not so fast ... (usb virtualization)

aeickho, 2016-09-25:
> @sijskes  @aeickho no, the dmesgtail on d13 works. but slows down gui, considerable.

you can reduce the slowness of md380 while dmesgtail with add a sleep into  md380-tool -> dmesgtail()

roelandjansen, 2016-09-25:
e.g. you mean that if it's faster it would be come unstable. Ok. 

How long does your dump take? I could see if I can natively dump it in the system too and see if that makes any difference in time.

aeickho, 2016-09-25:
@roelandjansen 

`time ./md380-tool spiflashdump file.bin`  (no keys are pressed) 

- on s13.020 (gps) after 12 sec no time update .. real    1m47.499s .. can't switch power off 
- on d13.020 real   1m48.584s... radio is working (power of is working)

`time ./md380-tool spiflashdump file.bin`  (->menu->back (repeats x times ) 
- on s13.020 not tested
- on d13.020 after 4 repeats - real    1m46.165s (dump is complied) - no time update .. can't switch power off

roelandjansen, 2016-09-25:
ok,

roeland@md380:~/src/d$ time ../md380tools/md380-tool spiflashdump dumped 
Dumping SPI Flash to 'dumped'.

real    1m53.467s
user    0m7.420s
sys     0m1.852s

radio stayed responsive. I did it just one time. I am backing up the vm at this moment and will try and see repeatedly if I also see this bevaviour. I did a single shot and that one worked OK.

travisgoodspeed, 2016-10-28:
Self assigning to take care of this over the weekend.  I'll spin up a few related issues to migrate the features that require fast polling away from dmesg.