travisgoodspeed/md380tools #303: Windows 10 Bash

paulomartinspt, 2016-09-14:
Tried it for the first time, installed all dependencies (i think) and got to this part:

```
Merging an applet.
Loading symbols from main.img.sym
Inserting a stub hook at 0800c72e to 0809c7e5.
Merging main.img into experiment.img at 0809b000
../md380-fw --wrap experiment.img experiment.bin
DEBUG: reading "experiment.img"
INFO: base address 0x800c000
INFO: length 0xf3000
DEBUG: writing "experiment.bin"
../md380-dfu upgrade experiment.bin
No module named core
make[1]: *** [flash] Error 1
make[1]: Leaving directory `/mnt/c/Users/xxxxxx/md380tools/applet'
make: *** [flash] Error 2
```

Any ideas on how to fix it?
Any data i can send to help?
Anyone has the process working on a Windows 10 machine through the built-in bash?

travisgoodspeed, 2016-09-14:
It's a libusb problem, and I don't think that libusb works in Windows Services for Linux.

Try running `make dist` and using the windows installer in the `md380-date\windows\` directory to flash the firmware update in the `md380-date` directory.

paulomartinspt, 2016-09-14:
Thanks for the fast reply, will do @travisgoodspeed 
This bash under Windows sure eases the process, let's hope we can get it working somehow.

paulomartinspt, 2016-09-14:
It worked @travisgoodspeed but the screen became all messy with the date and time flashing.

MD380 (old)
I used 
md380tools\md380tools-2016-09-14\firmware-2016-09-14-OLD.bin

Radio info:
MD380Tools Ver: 2016-09-14

travisgoodspeed, 2016-09-14:
We're under heavy development right now.  Do you see the same problem with the NoGPS variant?

Could you post a video or screenshot here?

paulomartinspt, 2016-09-14:
Yes, tried both @travisgoodspeed 
See: http://sendvid.com/7xxoz295

paulomartinspt, 2016-09-14:
That's strange @travisgoodspeed 
Tried to flash your latest FW compiled here: https://pd0zry.nl/md380-fw/ under Windows and i got the same problem. Gonna boot the Linux machine and do it again like i always did. Even last night i updated both FW and users.csv without any issue, it's the first time i see the screen flick since i use your FW.

paulomartinspt, 2016-09-14:
Just tried it on Linux with last night's copy of github, perfect.
Gonna delete the folder, download the rep, install again and see what happens.

aeickho, 2016-09-14:
you have seen #300 ... there was a fw for 1 day up stram that’s only  runs on d13.020

paulomartinspt, 2016-09-14:
It's today's copy @travisgoodspeed !!
After downloading the rep and compiling/flashing, i have the same problem, flicking screen!
That's why both my Windows process, the img download from that site or the Linux process give the same result. If i flash the copy of the repository from last night, it solves it.

aeickho, 2016-09-14:
an the version info on power on?  enable with utils->radio settings->introscreen-> char string

paulomartinspt, 2016-09-14:
@aeickho 
md380tools
by kk4vcz
and friends
@ d13.020
b879c8b
2016-09-14

aeickho, 2016-09-14:
disable debug on md380tools menu...  the flicking screen is for development .

paulomartinspt, 2016-09-14:
LOL. That's it.
So today's copy brings it active by default? Fixed of course.... thanks @aeickho 

Now, my only fight is with Windows 10 "linux mode".
I work with a lot of Windows machines and they all have bash active now.

paulomartinspt, 2016-09-14:
I know that this is not what i originally reported but it's just so i don't open another thread just for this.
@aeickho, can you please tell me what's that blue symbol that i have blinking on the right side of the speaker? Thanks!

https://s16.postimg.org/70qyz7avp/2016_09_14_23_18_23.jpg

aeickho, 2016-09-14:
thats shows promiscuous / monitor mode enabled (md380tools menu)  ... the hook is not very good ... the flickering is not nice ... but we don't know a better  hook position yet

mach327, 2016-09-15:
monitor mode icon flickering: #308

sijskes, 2016-09-15:
remaining issues is opened as #308

closed 2016-09-15 by sijskes