travisgoodspeed/md380tools #308: Promiscuous icon ('eye') blinks when idle

jruys, 2016-09-15:
The 'eye' icon in the idle screen flickers, like it's overwritten by something and re-drawn again.

When tuned to an active call (DMR active conversation or FM repeater) it appears stable.

sijskes, 2016-09-15:
patches are welcome.

sijskes, 2016-09-15:
this is a known issue, and no solution currently exists. keeping open for others.

jruys, 2016-09-18:
Is it possible to create a #define to disable the eye icon altogether? That way I can still use promiscuous but without an indicator. If someone can comment which module is responsible for drawing the icon happy to look into this.

Other option would be to only show it on RX, at which point the icon appears to be stable anyway.

sijskes, 2016-09-21:
@aeickho 
The eye icon does not flicker anymore. but background is now as reminant of the word 'menu'. 
i've identified the method that draws the statusline, the icon is in the middle of a blockfill zone, so the best thing would be to move the icon to the left or right. left is easiest.

sijskes, 2016-09-21:
@aeickho i even think the do_status_line gfx_drawbmp en gfx_blockfill can be intercepted by a special hook.

sijskes, 2016-09-21:
@aeickho i will start this now.

sijskes, 2016-09-22:
Also https://github.com/travisgoodspeed/md380tools/pull/344#issuecomment-248825478

sijskes, 2016-09-22:
Wait a while on how #345 is decided.

sijskes, 2016-09-29:
the promisc symbol is much easier on the eye right now.

closed 2016-09-29 by sijskes

xSmurf, 2016-10-23:
Am I right that this isn't fixed in the S release?

DataSquid, 2016-10-23:
It certainly blinks a lot on mine when there's no audio output.  The LED is often solid green at the same time as well.  I see similar behaviour with DMR on a Uniden scanner I have.  Where it will show activity on a timeslot but not decode any audio.  Could it be data?

sijskes, 2016-10-23:
@DataSquid verifyable with netmon. shall we carry this on in 

https://groups.google.com/group/md380tools

issues are not a forum. lets make some noise in google groups.