travisgoodspeed/md380tools #321: i don't like this particular function, could you fix it for me?

sijskes, 2016-09-18:
I really dislike this feature, and i think it should be easy to fix by one of the project members. 
Would you do the work for me?

sijskes, 2016-09-18:
The best way to get things fixed, is by doing the fix yourself, and do a pullrequest to get the fix integrated.

travisgoodspeed, 2016-09-18:
Pretty sure this is a duplicate of prior issues. :)

der-stefan, 2016-09-21:
@sijskes I can see your point here, but reality is different. Not everyone is able to provide code, let it be because of missing knowledge or simply due to limited time. However, new ideas and bug reports are also useful for this project. So calm down and kindly close this issue.

sijskes, 2016-09-21:
@der-stefan rather presumptuous are you?

sijskes, 2016-09-21:
@der-stefan "So calm down and kindly close this issue." it sounds like ordering somebody around. are you?

sijskes, 2016-09-21:
To others: good bug reports are always appreciated. Ideas too.

If you are not happy with how a certain feature is implemented, or dont like the quality of the implementation, there is no other option than to participate constructively in this project, and in the end this means, doing the hard work.

travisgoodspeed, 2016-09-23:
I've opened #370 to aggregate the ideas for new features which don't have active developer commitment, and I'll be merging older issues into that one this weekend.  Bug reports will retain their own issue numbers, as the duplicates don't hang around so long.

Closing this issue.

closed 2016-09-23 by travisgoodspeed