travisgoodspeed/md380tools #340: problem Netmon Function

PD1WP, 2016-09-21:
[image]
[image]
[image]

if got a problem with the Netmon funcion in your latest firmware update.
What does this feature because my handheld transceiver no longer work properly if netmon is turned on.
I have a few pictures added. 
Thanks for your support

sijskes, 2016-09-21:
have you tried with netmon: off, debug: off?

sijskes, 2016-09-21:
please try latest version, it is less intrusive when netmon is disabled.

PD1WP, 2016-09-21:
Yes..  I will flash the  latest version of your firmware. Thanks..  Is it possible to get a manual what netmon does? Thank you for your time.

sijskes, 2016-09-21:
it doesn't do much now. it shows some internal variables, channel name, tg name.
it helps a developer to see what is going on in the internals.
you can always look, but you need to understand the source code to make sense of it, and even then, it is still a mystery what most of the internal variables really do.

PD1WP, 2016-09-21:
Okay.. I just flash it and have the latest version.  The pictures shows my screen when I  enable the Netmon  function.  Do I need to enable also the debug mode?  The strange numbers disappears when I turn off the Netmon function. Thanks for taking the time to give me the answer on my questions..  (sorry.. For  English writing)  PD1WP Wilfred from Apeldoorn, 
[image]
[image]

 the Netherlands

sijskes, 2016-09-21:
no, you dont need debug mode. set it to disabled.

and yes, the strange numbers are from netmon. they ARE netmon.

netmon is a reference to a special mode that was included in nokia phones. not intended for normal users. it was a whole lot of very complicated numbers. 

if you dont like the strange numbers, disable netmon.

travisgoodspeed, 2016-09-23:
Closing this issue as the expected behavior of Netmon.

Perhaps we should enable the UsersCSV display by default, or segregate things like Netmon to a "developers only" menu, to lessen confusion?

closed 2016-09-23 by travisgoodspeed

sijskes, 2016-09-23:
i like the developer only menu remark.