travisgoodspeed/md380tools #345: Extra Top-Icons .. Top line full

aeickho, 2016-09-21:
the Top Icon line is full, on the position of the "eye" is also the gps (locked) icon

[image]

close this issues if we have an solution for this problem

see #77

aeickho, 2016-09-21:
- changing 2 ot more icon continuously 
- smaller icons
 - remove crypto icon ;)

m0ivq, 2016-09-22:
Maybe replace the 'shield' (encryption) icon with the 'eye' when in promiscuous mode, with a different colour for encrypted/unencrypted?

travisgoodspeed, 2016-09-22:
If we expect to add more than just the promiscuous icon, and we're already out of room, we'll have to either add another row or abother column.  Maybe the far right of the screen could hold a column of our icons, and the top row can hold the factory icons?

sijskes, 2016-09-22:
as we have most gfx_ primitives hooked now, we can implement a clipping rectangle to create a free zone anywhere.

sijskes, 2016-09-22:
the only thing is correctly finding screen state. the 'main' mode is 1 step behind the draw of the statusline.

paulomartinspt, 2016-09-29:
@travisgoodspeed i agree
I believe creating a new column would be better, we have a lot of free space, good idea

xSmurf, 2016-10-18:
Could it be moved to _20_ right now? That would make it appear on top of the _monitoring_ icon, which is unused in digital mode... (in fact, it would be even nicer if the promiscuous mode used the same monitoring/squelsh open icon, imho).

xSmurf, 2016-10-21:
After some testing, it appears 20 isn't ideal on non-gps model (D release). So I would suggest we leave it at 65 for D release and move it to 20 for S release.

DL2MF, 2016-11-09:
Since build #62c5c3a this has been fixed, with D13.020 the icon remains in the old place, on GPS models with S13.020 it moved to pos. 20 (right from RSSI).

closed 2016-11-09 by sijskes