travisgoodspeed/md380tools #361: make flashdb failed with md390 no gps

ZeroChaos-, 2016-09-23:
zero@gato md380tools % sudo make flashdb                                                                                      (git)-[master] 
cd db && make
make[1]: Entering directory '/home/zero/development/md380tools/db'
make[1]: Nothing to be done for 'update'.
make[1]: Leaving directory '/home/zero/development/md380tools/db'
iconv -c -f UTF-8 -t ascii//TRANSLIT db/users.csv | cut -d',' -f1-3,5-6 | sed 's/,\s+/,/g' > data.csv
wc -c < data.csv > data
cat data.csv >> data
./md380-tool spiflashwrite data 0x100000
SPI Flash ID: ef 40 18
W25Q128FV 16MByte
erase 2139942 bytes @ 0x100000
flashing 2139942 bytes @ 0x100000
reboot radio now
Exception usb.core.USBError: USBError(19, u'No such device (it may have been disconnected)') in <bound method Device.__del__ of <DEVICE ID 0483:df11 on Bus 002 Address 018>> ignored

(the radio reboots itself and does not appear to be able to identify any of the users programmed into the users.csv)

travisgoodspeed, 2016-09-23:
We really need more information, neighbor.  An error message without context isn't enough to help you.

1. Which operating system are you using?
2. Does 'make flash' work?
3. Which radio firmware version are you running?
4. Did it ever work for you in the past?

73 from Pizza Rat City,
-Travis

m0ivq, 2016-09-23:
The exception message is normal when the radio reboots itself. Have you enabled the users.csv option in the md380tools menu of the radio?

roelandjansen, 2016-09-23:
as Simon said -- this is to be expected after flashing as the radio restarts -- the disconnect is normal.
Be sure that you have enabled users.csv and disabled netmon

travisgoodspeed, 2016-09-23:
Closing this as the behavior is normal.

closed 2016-09-23 by travisgoodspeed

ZeroChaos-, 2016-12-15:
While this behavior is normal shouldn't the radio reboot be caught and handled so it doesn't appear to be a failure?