travisgoodspeed/md380tools #365: Squelch works not anymore

PD1WP, 2016-09-23:
I updated my Tytera md-390 to the latest version with Ubuntu And from that moment the Squelch does not work anymore on my analog channels. I also tried the version from PD0ZRY with the same results.  Now i have downloaded  the latest version from yesterday from the site PD0ZRY and now the Squelch works perfectly.

roelandjansen, 2016-09-23:
We also noticed wit the latest build we did. The version that works OK and still with most of the 
netmon features is here https://github.com/roelandjansen/md380tools/releases/tag/b0ba798 
and these builds are built directly after a merge has taken place.

sijskes, 2016-09-23:
@PD1WP thanks for your report. i didnt notice it during testing, too much focused on the dmr radio layer.

You can indeed find stable releases on the roelandjansen releases page. 

But i like people testing the 'unstable' releases as well. :)

roelandjansen, 2016-09-23:
I do too.

https://github.com/roelandjansen/md380tools/releases

is the place where all our builds are stored./ Older builds willl be deleted after some time. Also Once in a while we'll pick a stable release and publish it on the ham-dmr/hamdigitaal website

closed 2016-09-23 by sijskes

roelandjansen, 2016-09-23:
see

https://github.com/roelandjansen/md380tools/releases/tag/e8c3e56

with a working SQ.

commit d1ecedf7 (2016-09-23): fixed #365