travisgoodspeed/md380tools #370: Ye Grande Ol' Wish List for Christmas

travisgoodspeed, 2016-09-23:
A lot of folks have a lot of ideas for adding features, and while some of these are good ideas, they are a lot of work that fills the issue tracker and discourages us poor humble developers to no end.

Therefore, let's aggregate all of the wishlist items without active development to this one grand issue, temporarily closing them until (and unless) work begins on them.  Items under active development will keep their own issues and not be a part of this list.

- [ ] Status LED Signalling #90 
- [ ] Change DTMF Mapping #92 
- [ ] RSSI Value instead of Icon #93 

(I'll be completing this list over the weekend.)

mach327, 2016-09-24:
(I added a few to your list)

PD1LOI, 2016-10-01:
Scroll through the menu by holding up / down button! Also as an option.

trolfe, 2016-10-03:
Changing the channel via the USB, for one of the 16 in the zone?

vbradio, 2016-10-13:
Being able to use the Menu button , while the radio is actively receiving DMR would be nice.

travisgoodspeed, 2016-10-23:
@xSmurf Done.

kd8uxc, 2016-12-03:
I see a request similar to mine ( Backlight control #477).  Perhaps my request can be combined in the event the other request is taken on.  My slightly different way about addressing the issue may even satisfy the needs of the OP.

What about a way to turn the light off completely. Then have it only turn on when either the menu button or the back button is pressed.  When I am engaged in a QSO or listening I have no need for the light.  As the poster of request #477 stated it can be very annoying in low light conditions.  It's even more problematic (even dangerous) in a vehicle at night.  I have found myself covering the screen so I don't distract the driver.

scannermanuk, 2016-12-24:
Hi all,new to this but have interest in the md380, I have not purchased one yet. Could you get it to ignore the RAS ? And the ability to work along side the mototrbo basic privacy. Thanks

travisgoodspeed, 2016-12-25:
This is not a discussion forum.  Please keep discussions to the Google Group.

73 from Salt Lake City,
-KK4VCZ

kd8uxc, 2017-01-02:
**Extra Low Power Option for Openspot users/Personal repeater users.**  

Why?
1. Use the lowest power needed to get the job done.  It's just good practice.
2. Save battery.  

Hopefully a third setting but I would be extremely happy with the ability to simply decrease the setting of the low power option.  Perhaps a percentage setting with 2 or 3 options.  People are currently achieving this by messing with the factory calibration.

travisgoodspeed, 2017-01-03:
@kd8uxc Do you have information on what calibration changes they make?  I think we have enough info on the radio hardware to do a low power mode, but I'd like to see how it's currently done.

kd8uxc, 2017-01-03:
@travisgoodspeed 
None of this is my work.  It was posted by a member of the TYT-TYTERA yahoo group.

> Derating for TYT MD-380 / RETEVIS RT3 power level LOW approximately 100mW
> (for example, for operation with DV4mini stick)
> 1. Go to the directory where the code plug-software is stored (for me: C:\Program Files(x86)\Tytera\MD_380\SoftWare_EN
> 2. there you find the File setting.ini, open the file, select the parameter to test mode = 1 change in line 3 (orig .: testMode = 0),
> if it was not going: put (copy) the setting.ini file on desktop, modify, copy it back!
> 3. Connect and turn on radio with programming cable
> 4. Start CP software, and then press Ctrl-T or Strg-T, a window should open
> maybe it looks like this:
> 5. focus in this window to the vertical column 5
> 6. there you should looking for the value for TxLowPower (column 5) and a note the original value to possibly later restore to
> the original state.
> 7. Now halve this value and submit (in my case Value 74, now 37)
> 8. now restore the parameters to the device using the Write Button (IMPORTANT!)
> 9. Close the window by clicking Cancel, exit CPS and remove programming cable.
> The radio should now make in position LOW Power about 100mW, the power at high power remains unaffected.
> PLEASE DON'T Change any other parameters !!!, you could make the device unusable else!
> This modification IS AT YOUR OWN RISK !!
> 73

[_low-power-md-380.pdf](https://github.com/travisgoodspeed/md380tools/files/681778/_low-power-md-380.pdf)

dragstor, 2017-03-11:
Day / night mode would be great :)

If I'm not wrong, we could alter the screen colors depending on the time of day. 

Eg: during the morning - afternoon hours, background would be white with black text. 
From 6pm up to 7 AM, black/grey with appropriate text colors.

Possible or scifi? :)

DL2MF, 2017-08-29:
We already made some progress during the last months, the following features are available now:

- Mic Gain Option #127 - done
- Faster Contact Navigation #158 -done
- Last heard list (currently without GPS) #198 - done
- Feature: Talker Alias #681- done

In my developer branch already available:
- Promiscuous mode On/Off programmable on the side button would be handy as well. (Friday, April 14 2017)
- Toggle Mic gain with side button

Swenum, 2018-01-15:
Listening to MotoTRBO RAS repeaters. We use it on a dozen repeaters. If necessary, I can record packets from the air.

mach327, 2018-05-30:
GPS/location data out #834

adron51, 2020-03-25:
it would be nice if the option to choose display background from biue with white letters to black with white letters was so people that poor eyesite could choose what fits them best

KD4Z, 2020-03-25:
@adron51   Since you are using the firmware built by my Toolkit, you should investigate the tweaks menu.  There, you will find what you seek:
    touch ~/blackfontrxscreen.enable
run glv and then flash again for the new firmware to show in black font.