travisgoodspeed/md380tools #380: objects may fail ?

ispkorte, 2016-09-24:
if i make "make clean" and "make flash"  i get some "objects may fail" on my Linux Mint System

/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: main.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: printf.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: dmesg.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: usb.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: gfx.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: stm32f4xx_it.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: system_stm32f4xx.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: addl_config.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: spiflash.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: menu.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: rtc_timer.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: beep_process.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: usersdb.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: os.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: aes.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: ambe.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: dmr.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: display.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: console.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: util.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: netmon.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: debug.o uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: lib/libstm32f4.a(stm32f4xx_gpio.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
/usr/lib/gcc/arm-none-eabi/4.8.2/../../../arm-none-eabi/bin/ld: warning: lib/libstm32f4.a(stm32f4xx_rcc.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail

sijskes, 2016-09-24:
Could you try the last version? i've fixed this today.

ispkorte, 2016-09-24:
yes , it is the last one
same on RT3 and RT8

roelandjansen, 2016-09-24:
I did not see this, Simon with the latest commits.

roelandjansen, 2016-09-24:
Have you tried a git pull?

sijskes, 2016-09-24:
Could you clone the repo again in a clean directory, and try again?

ispkorte, 2016-09-25:
sure ;)

-- 

mit freundlichen Gruessen / kind regards

Joerg Korte

Internet Business Consultant

	Vor dem Drucken dieser E-Mail denken Sie bitte
	 an den Schutz der Natur und unserer Umwelt!

-----------------------------------------------------------------
-- Linux wird nie das meist installierte Betriebssystem sein,   --
-- wenn man bedenkt, wie oft man Windows neu installieren muß! --
-----------------------------------------------------------------
--                       I S P - K O R T E                     --
-- Resellerservice of Web Server & Web Hosting & VoIP Systems  --
-----------------------------------------------------------------
Support Adresse Deutschland:

         ISP-Korte
         -Joerg Korte-
         Idafehn Nord 115
         26842 Ostrhauderfehn
-----------------------------------------------------------------
Service : +49 9005 102 895 0 (1.99 Eur/min)°
Telefon : +49 180-3118877-00 (9cpm)°
Fax : +49 180-3118877-10 (9cpm)°

Italy	+39 02 00 61 21 34
Germany	+49 49 61 98 54 58

° aus dem netz der DTAG / mobil auch mehr
-----------------------------------------------------------------
UId:  DE248861249
EORI: DE 406 2631 3424 3657
B-Netz-A: 	RegNr 	08/297
	ID 	108/177
-----------------------------------------------------------------
Homepage:
         http://www.isp-korte.de
                 unsere Resellerplatform
                 Webhosting preiswert
Messenger:
	ICQ     : 317316709
-----------------------------------------------------------------

ispkorte, 2016-09-25:
now it was  run whitout fail

perfect

-- 

mit freundlichen Gruessen / kind regards

Joerg Korte

Internet Business Consultant

	Vor dem Drucken dieser E-Mail denken Sie bitte
	 an den Schutz der Natur und unserer Umwelt!

-----------------------------------------------------------------
-- Linux wird nie das meist installierte Betriebssystem sein,   --
-- wenn man bedenkt, wie oft man Windows neu installieren muß! --
-----------------------------------------------------------------
--                       I S P - K O R T E                     --
-- Resellerservice of Web Server & Web Hosting & VoIP Systems  --
-----------------------------------------------------------------
Support Adresse Deutschland:

         ISP-Korte
         -Joerg Korte-
         Idafehn Nord 115
         26842 Ostrhauderfehn
-----------------------------------------------------------------
Service : +49 9005 102 895 0 (1.99 Eur/min)°
Telefon : +49 180-3118877-00 (9cpm)°
Fax : +49 180-3118877-10 (9cpm)°

Italy	+39 02 00 61 21 34
Germany	+49 49 61 98 54 58

° aus dem netz der DTAG / mobil auch mehr
-----------------------------------------------------------------
UId:  DE248861249
EORI: DE 406 2631 3424 3657
B-Netz-A: 	RegNr 	08/297
	ID 	108/177
-----------------------------------------------------------------
Homepage:
         http://www.isp-korte.de
                 unsere Resellerplatform
                 Webhosting preiswert
Messenger:
	ICQ     : 317316709
-----------------------------------------------------------------

closed 2016-09-25 by sijskes