travisgoodspeed/md380tools #386: integration of version specific makefiles

sijskes, 2016-09-24:


travisgoodspeed, 2016-09-26:
Windows does run GNU Make, but often there is a space in `make`'s path just to mess with you.  BSD runs BSD make, of course, but has `gmake` as a separate command.

Let's first push a draft of the new Makefile, then quickly have some Windows folks try it and adjust the documentation before making it the default.

sijskes, 2016-09-28:
made mods on applet/Makefile. saved original for quick revert.
validated D02 build by flashing and using.

sijskes, 2016-10-01:
are there any issues detected?

travisgoodspeed, 2016-10-01:
Works fine on Mac, but we don't yet have confirmation for GNUWin32 or Cygwin.  (Windows Services for Linux should be fine.)

Is anyone using BSD for this project?

m0ivq, 2016-10-01:
I'll check on GNUWin32 & Cygwin tomorrow.

m0ivq, 2016-10-02:
Works ok under GNUWin32, not yet tested on Cygwin.

travisgoodspeed, 2016-10-02:
I think this is responsible for 'make dist' being broken in GNU Make 3.81 on OS X.  GNU Make 4.1 on the same platform works without trouble.

Fix before closing the issue:
- [ ] GNU Make 3.81 on OS X

sijskes, 2016-10-02:
@travisgoodspeed do you have more info? is there a build environ where i can see the errors myself?

m0ivq, 2016-10-02:
I didn't test properly under Windows! 'make dist' gives the following error:

`C:/Progra~1/GnuWin32/bin/make -C applet --eval=FW=D13_020 all`
`C:/Progra~1/GnuWin32/bin/make: unrecognized option ```--eval=FW=D13_020'`

travisgoodspeed, 2016-10-02:
GNU Make 3.81 is the version shipped with XCode, so any Mac will work as a build environment.

@m0ivq I see the same error that you do.

```
/Applications/Xcode.app/Contents/Developer/usr/bin/make -C applet --eval=FW=D13_020 all  
/Applications/Xcode.app/Contents/Developer/usr/bin/make: unrecognized option `--eval=FW=D13_020'
```

sijskes, 2016-10-02:
@travisgoodspeed i have it converted to -D if P1 is merged please test again.

sijskes, 2016-10-02:
please test after merge. i've removed the --eval from P1

sijskes, 2016-10-02:
@travisgoodspeed i've done the merge. went back to old style defines. could you test again?

travisgoodspeed, 2016-10-02:
I'll be able to test in an hour or so.  @m0ivq might be able to test sooner, as he can reproduce the same issue.

m0ivq, 2016-10-02:
@sijskes now works on Windows (make 3.81) though I need to update the Windows Instructions to install 'zip.exe'

Make 4.1 under windows has other issues but may be down to my installation.

sijskes, 2016-10-02:
@m0ivq make4.1 windows, what kind?

m0ivq, 2016-10-02:
@sijskes I get the following but haven't had time to look into it yet:

`C:/Progra~1/GnuWin32/bin/make -C lib
make[2]: Entering directory 'C:/Users/post2/Desktop/git/travisgoodspeed/md380tools/applet/lib'
arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o misc.o src/peripherals/misc.c
arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o stm32f4xx_dma.o src/peripherals/stm32f4xx_dma.c
arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o stm32f4xx_rcc.o src/peripherals/stm32f4xx_rcc.c
arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o stm32f4xx_adc.o src/peripherals/stm32f4xx_adc.c
arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o stm32f4xx_exti.o src/peripherals/stm32f4xx_exti.c
arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o stm32f4xx_rng.o src/peripherals/stm32f4xx_rng.c
arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o stm32f4xx_can.o src/peripherals/stm32f4xx_can.c
arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o stm32f4xx_flash.o src/peripherals/stm32f4xx_flash.c
arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o stm32f4xx_rtc.o src/peripherals/stm32f4xx_rtc.c
arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o stm32f4xx_crc.o src/peripherals/stm32f4xx_crc.c
arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o stm32f4xx_fsmc.o src/peripherals/stm32f4xx_fsmc.c
arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o stm32f4xx_sdio.o src/peripherals/stm32f4xx_sdio.c
arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o stm32f4xx_cryp_aes.o src/peripherals/stm32f4xx_cryp_aes.c
arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o stm32f4xx_gpio.o src/peripherals/stm32f4xx_gpio.c
arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o stm32f4xx_spi.o src/peripherals/stm32f4xx_spi.c
arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o stm32f4xx_cryp.o src/peripherals/stm32f4xx_cryp.c
arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o stm32f4xx_hash.o src/peripherals/stm32f4xx_hash.c
arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o stm32f4xx_syscfg.o src/peripherals/stm32f4xx_syscfg.c
arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o stm32f4xx_cryp_des.o src/peripherals/stm32f4xx_cryp_des.c
arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o stm32f4xx_hash_md5.o src/peripherals/stm32f4xx_hash_md5.c
arm-none-eabi-gcc -g -O2 -Wall -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -ffreestanding -nostdlib -Iinc -Iinc/core -Iinc/peripherals -fshort-wchar -c -o stm32f4xx_tim.o s [... cut ...]
arm-none-eabi-ar: creating libstm32f4.a
make[2]: Leaving directory 'C:/Users/post2/Desktop/git/travisgoodspeed/md380tools/applet/lib'
g++    -c -o main.o src/main.c
process_begin: CreateProcess(NULL, g++ -c -o main.o src/main.c, ...) failed.
make (e=2): The system cannot find the file specified.
<builtin>: recipe for target 'main.o' failed
make[1]: *** [main.o] Error 2
make[1]: Leaving directory 'C:/Users/post2/Desktop/git/travisgoodspeed/md380tools/applet'
makefile:22: recipe for target 'applets' failed
make: *** [applets] Error 2`

sijskes, 2016-10-02:
@m0ivq is looks like it tries to start the stock g++ compiler. but stranger still why does it compile main.o with the g++ compiler? is the main.c actually main.C ?

travisgoodspeed, 2016-10-02:
@sijskes Confirmed that the fix works for GNU Make 3.81 on OS X.

sijskes, 2016-10-04:
@m0ivq we can try by overriding the .C rule with the same as .c

sijskes, 2016-10-27:
implemented

closed 2016-10-27 by sijskes