travisgoodspeed/md380tools #395: md380-tool calllog

ispkorte, 2016-09-28:
if i start **md380-tool calllog** and at the moment a call come in , radio will hangout
see at picture screen after 2 min make off radio (its not OFF , it is ON)

[image]

sijskes, 2016-09-28:
Can you tell us more? Which version? Does it still occur with the latest? Could you try the stable release @ https://github.com/roelandjansen/md380tools/releases/tag/66d3e86

ispkorte, 2016-09-28:
its a RT8 S13 - stable - from 2016-09-27

sijskes, 2016-09-28:
is it reproducable? 

does it still occur with the lastest version: https://github.com/roelandjansen/md380tools/releases/tag/e08b733

sijskes, 2016-09-28:
So to be clear, the volume knob was turned in the off position?

ispkorte, 2016-09-28:
radio was fail after a call on 232 was come in , 5 min nothing doing i was make OFF but radi was not off ; after 2min i have open accu for off

ispkorte, 2016-09-28:
i have had **md380tools-66d3e86** inside RT8 
start **./md380-tool calllog** but nothing comes 
in some times a call go in - display stop (see picture) and ratio was freez
also in the daily version from today 

i have no idea about fail at my side or not

sijskes, 2016-09-28:
@ispkorte  i don't think it is a fail at your side. 
did the problem repeat itself any other time?

ispkorte, 2016-09-28:
at 66d3e86 i can repeat fail ; now i have the new 0666ab8 from 1PM(UTC) and no fail
but i get no line at **./md380-tool calllog**

sijskes, 2016-09-28:
@aeickho is calllog ever ported to D13? i see memory constants in md380-tool.

travisgoodspeed, 2016-09-28:
The calllog tool is an ugly, ugly hack from the early days of this project.  We should make our own buffer for recent calls from our C hooks instead.

ispkorte, 2016-09-28:
ok fellow , close for now until you have time back for restart

closed 2016-09-28 by ispkorte