travisgoodspeed/md380tools #403: MD380 / RT3 support D-Star and C4FM ?

ispkorte, 2016-09-29:
i found (long time ago) a web link from **VA3XPR**  [http://www.va3xpr.net/hams-hack-110-dmr-radio-allow-support-d-star-p25-system-fusion/] that MD/RT hack will support **D-STAR** and/or **C4FM** . it sound nice and interest , but i do not found any other info about that mode for it.

is it wrong or right ?

73 de DD1GO / HS0ZKJ

paulomartinspt, 2016-09-29:
Since MD380 is an "sdr radio", i don't see why not. There can be some issues with the hardware limitations but let's see what the guys say about this, mainly @travisgoodspeed

travisgoodspeed, 2016-09-29:
The MD380 is not an SDR, and we can't do Fusion or DStar without a raw symbol mode that the radio baseband chip (C5000) doesn't seem to offer. It's a shame, because all the other parts of those protocols could be done, we just can't get the frames on and off the air.

If you happen to be fluent in the embedded systems vocabulary of Chinese, I can point you toward the documentation for a second opinion. (Not being snarky, that's what the C5000 is documented in.)

Closing the issue.

Cheers from Philly,
-Travis

closed 2016-09-29 by travisgoodspeed