travisgoodspeed/md380tools #409: Promiscuous mode not working

vbradio, 2016-09-30:
Promiscuous mode not working ( version 0449571 ) and possible some older version? Can somebody verify

dc7jzb, 2016-09-30:
Made a firmware today, and promiscious mode don't work for me too.
Running MD380 with old Vocoder.

travisgoodspeed, 2016-10-01:
I've confirmed that the bug is real and affects only D13.020 (Non-GPS) firmware.  As S13.020 (GPS) doesn't yet run on non-GPS hardware, you should flash 2.032 (OLD) as a temporary fix for promiscuous mode.

Self assigning.  I've been away from the codebase for a few weeks, and it'll be good to get my hands dirty again.

travisgoodspeed, 2016-10-01:
The break is caused by hooks for `OSMboxPost()`, which is hooked first to `dmr_apply_squelch_hook` and then re-hooked to a different handler at `OSMboxPost_hook`.

This patch disables the second set of OSMboxPost hooks, which restores the promiscuous mode but likely causes trouble for some of @sijskes's recent work.

```
iMac% git diff
diff --git a/applet/merge_d13.020.py b/applet/merge_d13.020.py
index 77698e2..a4f8f4c 100644
--- a/applet/merge_d13.020.py
+++ b/applet/merge_d13.020.py
@@ -897,8 +897,8 @@ if __name__== '__main__':
         0x0808ee3c ,
         0x0808eeea ,
         ];
-    for adr in OSMboxPost_hook_list:
-        merger.hookbl(adr,sapplet.getadr("OSMboxPost_hook"),0);
+    #for adr in OSMboxPost_hook_list:
+    #    merger.hookbl(adr,sapplet.getadr("OSMboxPost_hook"),0);
 
 ###    #Throwaway hook to see if adr is called.
 ###    #merger.hookstub(0x0803f03c,
iMac% 
```

Assigning to @sijskes for a proper fix, as he is actively working on the hooks in question.

commit 71c02f1c (2016-10-01): fixed promisc
hopefully will keep working mbox post hook re-enabled next time
closes #409