travisgoodspeed/md380tools #424: Blank display issue after flashing MD380

on8vm, 2016-10-02:
After flashing my MD380 with the latest version, I occasionally have a blank display.

When the display becomes blank (just blue backlight), RX and TX work well.
I cannot reproduce the problem: it happens when just listening or when commuting memories.

Here below the build log (from http://pd0zry.nl/md380-fw/) of the firmware version I have used.

73, ON8VM

firmware-20161002-180001-1a1f1d15

rm -f data data.csv	
cd patches/2.032 && make clean
make[1]: Entering directory '/home/zarya/md380tools/patches/2.032'
rm -f *~ *.bin *.img
make[1]: Leaving directory '/home/zarya/md380tools/patches/2.032'
cd patches/d13.020 && make clean
make[1]: Entering directory '/home/zarya/md380tools/patches/d13.020'
rm -f *~ *.bin *.img
make[1]: Leaving directory '/home/zarya/md380tools/patches/d13.020'
cd firmware && make clean
make[1]: Entering directory '/home/zarya/md380tools/firmware'
rm -rf bin unwrapped
make[1]: Leaving directory '/home/zarya/md380tools/firmware'
cd applet && make clean
make[1]: Entering directory '/home/zarya/md380tools/applet'
make -C lib clean
make[2]: Entering directory '/home/zarya/md380tools/applet/lib'
rm -f misc.o stm32f4xx_dma.o stm32f4xx_rcc.o stm32f4xx_adc.o stm32f4xx_exti.o stm32f4xx_rng.o stm32f4xx_can.o stm32f4xx_flash.o stm32f4xx_rtc.o stm32f4xx_crc.o stm32f4xx_fsmc.o stm32f4xx_sdio.o stm32f4xx_cryp_aes.o stm32f4xx_gpio.o stm32f4xx_spi.o stm32f4xx_cryp.o stm32f4xx_hash.o stm32f4xx_syscfg.o stm32f4xx_cryp_des.o stm32f4xx_hash_md5.o stm32f4xx_tim.o stm32f4xx_cryp_tdes.o stm32f4xx_hash_sha1.o stm32f4xx_usart.o stm32f4xx_dac.o stm32f4xx_i2c.o stm32f4xx_wwdg.o stm32f4xx_dbgmcu.o stm32f4xx_iwdg.o stm32f4xx_dcmi.o stm32f4xx_pwr.o libstm32f4.a
make[2]: Leaving directory '/home/zarya/md380tools/applet/lib'
rm -f *.img *.lst *.bin *.elf *.hex *.o *~ *.sym src/*~ src/version.h
make[1]: Leaving directory '/home/zarya/md380tools/applet'
rm -f *~ *.pyc
cd firmware && make all
make[1]: Entering directory '/home/zarya/md380tools/firmware'
mkdir -p bin
unzip -p dl/D002.024.zip "$(cat firmware_files.txt | grep "^D002.024" | cut -f3)" >bin/D002.024.bin
unzip -p dl/D002.025.zip "$(cat firmware_files.txt | grep "^D002.025" | cut -f3)" >bin/D002.025.bin
unzip -p dl/D002.026.zip "$(cat firmware_files.txt | grep "^D002.026" | cut -f3)" >bin/D002.026.bin
unzip -p dl/D002.027.zip "$(cat firmware_files.txt | grep "^D002.027" | cut -f3)" >bin/D002.027.bin
unzip -p dl/D002.030.zip "$(cat firmware_files.txt | grep "^D002.030" | cut -f3)" >bin/D002.030.bin
unzip -p dl/D002.032.zip "$(cat firmware_files.txt | grep "^D002.032" | cut -f3)" >bin/D002.032.bin
unzip -p dl/D002.034.zip "$(cat firmware_files.txt | grep "^D002.034" | cut -f3)" >bin/D002.034.bin
unzip -p dl/D003.008.zip "$(cat firmware_files.txt | grep "^D003.008" | cut -f3)" >bin/D003.008.bin
cp dl/D003.020.bin bin/D003.020.bin
cp dl/S003.012.bin bin/S003.012.bin
cp dl/D013.009.bin bin/D013.009.bin
cp dl/D013.014.bin bin/D013.014.bin
cp dl/D013.020.bin bin/D013.020.bin
cp dl/S013.012.bin bin/S013.012.bin
cp dl/S013.020.bin bin/S013.020.bin
cp dl/MD280.002.018.bin bin/MD280.002.018.bin
cp dl/MD446.D003.020.bin bin/MD446.D003.020.bin
cp dl/MD446.D013.020.bin bin/MD446.D013.020.bin
mkdir -p unwrapped
../md380-fw --unwrap bin/D002.024.bin unwrapped/D002.024.img
DEBUG: reading "bin/D002.024.bin"
INFO: base address 0x800c000
INFO: length 0xf4000
DEBUG: writing "unwrapped/D002.024.img"
../md380-fw --unwrap bin/D002.025.bin unwrapped/D002.025.img
DEBUG: reading "bin/D002.025.bin"
INFO: base address 0x800c000
INFO: length 0xf2800
DEBUG: writing "unwrapped/D002.025.img"
../md380-fw --unwrap bin/D002.026.bin unwrapped/D002.026.img
DEBUG: reading "bin/D002.026.bin"
INFO: base address 0x800c000
INFO: length 0xf2800
DEBUG: writing "unwrapped/D002.026.img"
[... 439 lines cut ...]
arm-none-eabi-gcc -std=gnu99 -g -O2 -Wall -Tapplet_d13.020.ld -mlittle-endian -mthumb -mcpu=cortex-m4 -mthumb-interwork -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fshort-wchar -Wa,-a=main.elf -DFW_D13_020	 -Iinc -Ilib -Ilib/inc -I. -Ilib/inc/core -Ilib/inc/peripherals -D "MD380_d13_020"   -Wl,--no-wchar-size-warning -Wl,-Map,main.map main.o printf.o dmesg.o gfx.o stm32f4xx_it.o system_stm32f4xx.o addl_config.o spiflash.o usb.o menu.o rtc_timer.o beep_process.o usersdb.o os.o aes.o ambe.o dmr.o display.o console.o util.o netmon.o debug.o keyb.o radiostate.o lib/startup_stm32f4xx.s -o main.elf -Llib -lstm32f4 -Xlinker --just-symbols=src/symbols_d13.020
arm-none-eabi-objcopy -O binary main.elf main.img
arm-none-eabi-objdump -t main.elf >main.img.sym
cd ../patches/d13.020 && make all
make[2]: Entering directory '/home/zarya/md380tools/patches/d13.020'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/zarya/md380tools/patches/d13.020'
cp ../patches/d13.020/patched.img ./experiment.img
# use a fresh experiment.img
cd ../patches/d13.020 && make all
make[2]: Entering directory '/home/zarya/md380tools/patches/d13.020'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/zarya/md380tools/patches/d13.020'
cp ../patches/d13.020/patched.img ./experiment.img
python2 merge_d13.020.py experiment.img main.img 0x0809b000
Merging an applet.
Loading symbols from main.img.sym
Inserting a stub hook at 0800c72e to 0809c915.
Merging main.img into experiment.img at 0809b000
../md380-fw --wrap experiment.img experiment.bin
DEBUG: reading "experiment.img"
INFO: base address 0x800c000
INFO: length 0xf3000
DEBUG: writing "experiment.bin"
make[1]: Leaving directory '/home/zarya/md380tools/applet'

sijskes, 2016-10-02:
could you please try the stable build at https://github.com/roelandjansen/md380tools/releases/latest/ and report back?

on8vm, 2016-10-02:
Thanks Simon,

just flashed with f62053f-MD380-experimental-firmware.bin you suggested.

I had the blank display issues already twice, by simply browsing the memories in a zone.

FYI I have enable users.csv (working fine!) and rogerbeep, while debug, promiscuous, experimental and devonly are disabled.

KR,
Vito

sijskes, 2016-10-02:
@on8vm thanks for reporting the settings. could you explain what you mean with browsing the memories in a zone means? do you do changing zones? or contacts?

sijskes, 2016-10-02:
@on8vm i just fixed a buffer overrun in user.csv popup. can you test without users.csv popup, or use the pre-release https://github.com/roelandjansen/md380tools/releases/tag/6d5a5f6

on8vm, 2016-10-03:
@sijskes I flashed my md380 with the latest fw you provided.
The blank display issue appeared again, with users.csv enabled.
FYI after flashing it with the new fw, I re-wrote the SPI flash with the newest users.csv (from a raspberry pi, using 2016-05-27-raspbian-jessie-lite).

I have now disabled users.csv feature and I'll test for a while.

tnx, on8vm

sijskes, 2016-10-04:
what do you mean with browsing the memories? turning the channel selector?

on8vm, 2016-10-04:
Yes, just this

on8vm, 2016-10-04:
Tested the fw at URL https://github.com/roelandjansen/md380tools/releases/tag/6d5a5f6: I have the blank display issue with and without enabling users.csv.

I have also noticed that the display changes color: yellow on the bottom part and reddish/brown on the main area. Switching off and on, it becomes blue again as usual...

It's not the first experimental fw I try on my (I've already tried 3 or 4 previous versions in June and July 2016) but it's first time I encounter this strange behavior of the display.

on8vm, 2016-10-04:
[image]

on8vm, 2016-10-04:
[image]

sijskes, 2016-10-04:
Yes. it is very strange. i've not seen this behaviour before. I will keep this issue open for others to comment. 
Maybe is your codeplug that somehow causes this behaviour. To exclude any hardware failure can you try on a other ham's md380 with the same users.csv codeplug and firmware version.

on8vm, 2016-10-06:
@sijskes 
I have put back the old firmware (firmware-20160727-170001-b17ce768.bin) and the blank and changing color display issues don't appear anymore...
I'm back to the "post-it" version do users.csv info display... I liked the full screen.

But now the display behavior is definitely stable...

KR,
ON8VM

on8vm, 2016-10-07:
Hello,

I'm uploading the codeplug I'm using now and in the past.
The one I was using with the latest firmwares, that caused the blank (and strange colors) screen issue on my MD380.
The same I'm using now with the older (27/7/2016) firmware (firmware-20160727-170001-b17ce768.bin).
 
(rename the file to .zip)

[20161004_ON8VM_TYT380_PMR_Analogue.txt](https://github.com/travisgoodspeed/md380tools/files/516935/20161004_ON8VM_TYT380_PMR_Analogue.txt)

sijskes, 2016-10-07:
@on8vm can you try this last one?

on8vm, 2016-10-08:
@sijskes 
Hi Simon,

Have you prepared a specific build available at https://github.com/roelandjansen/md380tools/releases/tag with the latest change?

Or your change is part of the latest build that everyone can use?
ex. downloading the bin http://pd0zry.nl/md380-fw/firmware-20161008-160001-6a75f895.bin or via a "sudo make all flash"?

I play with firmwares but I'm not really an expert in software/firmware development...

TNX, on8vm

sijskes, 2016-10-08:
No the repo of @roelandjansen is not up to date yet, for testing your specific problem, the latest from pd0rzy is fine.

on8vm, 2016-10-09:
@sijskes 
I tried the build of firmware-20161008-220001-1ed9b3d9.bin (08-Oct-2016 22:02).

I have always the same problems: blank display and altered colors (I can live with the last issue... the first is quite annoying).

[image]
[image]
[image]

on8vm, 2016-10-09:
FYI the fw version I reverted to my md380 and that doesn't give any display problem:

[image]

mach327, 2016-10-09:
Using pd0zry's builds, the last firmware that does not exhibit this behavior on my VHF MD-380 is:
http://pd0zry.nl/md380-fw/firmware-20161008-130001-b8533dad.bin
and the first that does exhibit this behavior is:
http://pd0zry.nl/md380-fw/firmware-20161008-160001-6a75f895.bin

ik2cnf, 2016-10-09:
I acknowledge the same problem reported by mach327 on my UHF MD-380.

At start, screen is a white display but for channel number. Turning channels knob changes the number displayed, screen still white. (see pict 1)
Turning volume knob makes empty volume bar appear, but it remains on the sceen. (see pict 2)
Pressing menu button makes menu screen appear, but without icons. Pressing downkey button makes icon appear, one press down one icon. (see pict 3)
If you press exit key you never return to home page, last menu displayed still appears on screen.

I agree with mach37 that firmware-20161008-130001-b8533dad.bin from pd0rzy's builds is the latest working without problems.
[image]
[image]
[image]
[image]

sijskes, 2016-10-09:
@ik2cnf what settings do you use? debug? netmon? promisc?

sijskes, 2016-10-09:
@mach327 netmon display switching is far from perfect. ack.

sijskes, 2016-10-09:
a fix for this, is to disable netmon.

sijskes, 2016-10-09:
and also disable debug. netmon and debug are developer features. some incompletness is to be exptected.

ik2cnf, 2016-10-09:
I used Promiscuous, UserCSV and Mic Bargraph.
Debug disabled.
Experimental off, so I don't even know about netmon :-)

sijskes, 2016-10-09:
@ik2cnf when you startup, there is a blank screen? even no logo or text message?

sijskes, 2016-10-09:
do you have the commit hash from the working version and the first non-working?

ik2cnf, 2016-10-09:
@sijskes at startup I do see the logo and text msg. From then after home screen is as in picture 1 in my previous message.

I realized the problem this morning when I ugraded from last commit.
I don't remember when I upgraded before.

But I tried 3 builds from http://pd0zry.nl/md380-fw/

Just as @mach327, http://pd0zry.nl/md380-fw/firmware-20161008-130001-b8533dad.bin is the last working, next 2 http://pd0zry.nl/md380-fw/firmware-20161008-160001-6a75f895.bin and http://pd0zry.nl/md380-fw/firmware-S13.020-20161008-170001-dc61c974-buildlog.txt don't.

So last commit hash working should be  b8533dad and first not working 6a75f895

Sorry I can't help you with code, but I'm at your disposable for testing

sijskes, 2016-10-09:
@ik2cnf ok, can you double check your config when you have https://github.com/travisgoodspeed/md380tools/commit/6a75f8950a757e8e54a42366575094425d720f84 ?
this might be a bug in upgrading the config in spiflash.

ik2cnf, 2016-10-09:
@sijskes 
Simon please, what do you mean with "check your config"? Radio configuration?
Sorry but I'm newby to github, and the only way I know to upload an old firmware is by means of md380-dfu and bin in pd0zry repository.
So if you mean other tha radio menu configuration, please be more explicit I'd be glad to try

sijskes, 2016-10-09:
@ik2cnf i meant the settings like 'debug' 'netmon' 'experimental'

ik2cnf, 2016-10-09:
Fine!
now I cannot do a test, but as soon as I can I'll give a try with a plain configuration and report here
Thanks

on8vm, 2016-10-09:
On my side, I have also tried the last firmware that wasn't giving you display problems (20161008-130001-**b8533dad**).
I have always the blank screen issue. Even just after startup (that I see without problems).

In my case **the only enabled feature is users.csv**, all the rest (experimental, promiscuous, devonly) is disabled.

KR,
VM

roelandjansen, 2016-10-09:
also be aware that it is difficult for a start to support the builds other than what we provide in https://github.com/roelandjansen/md380tools/releases

The build that we provide are tested and we do pick stable releases and generally know their issues.

As @sijskes said -- disable debug, netmon, experimental stuff -- those are debugging/ engineering aids that are not expected to work for everyone (and having said that -- when those options are on, you are on your own -- generally we don't take bug reports on that).

Now that this is said -- could you load an official TYT firmware,  and then flash the latest stable release from the link above, with experimental/debug/netmon off? Also re-load the config file (inadvertedly being named "codeplug',  without any changes (!) and edit your DMR ID on the TYT itself. 

We need to have a clean start here, hence this request.

mach327, 2016-10-09:
Unfortunately, my VHF md-380 shows this bug, and I have not enabled Netmon - yet it acts as if I have.
I cannot even enter the menu to turn it off - the radio appears to crash, and reboots.

I will try a clean slate and see where that gets the VHF radio - will report back.

ik2cnf, 2016-10-09:
**Update**
I started from last working firmware b8533da and disabled all features in MD380tools menu.
Then I upgraded to 6a75f89, as requested by Simon.
This time at reboot I had **no** problem. I reintroduced, one by one, Promiscuous Mode, UsersCSV and Mic bargraph, each time rebooting the radio. I tried any possible combinations of the 3 features and I had no problem.
So I made a git pull, make clean and make all flash and now my MD380 is running last commit 4687f7c without troubles.
I do not know what kind of problem occurred, but I should recommend to restart with a tested firmware and a clean config in case of troubles.
I would like to learn what caused this strange behaviour.
Any hint appreciated
Thank you all

mach327, 2016-10-09:
I wrote a clean codeplug from `codeplugs/` to radio.
Then, I did the following procedure twice - once for the [latest stable release](https://github.com/roelandjansen/md380tools/releases/tag/e762fc9) and once for the `6a75f895` firmware.

1. Flash official Tytera firmware. Booted it and confirmed no issues.
2. Flash the new firmware. Booted it and confirmed issue.
3. Flash the last working firmware (well, last working pd0rzy build from above) - ensure netmon, debug, and all other md380tools features are disabled. Issue not present.
4. Flash the new firmware. Issue still exists.

Any firmware past '6a75f895' appears to make my VHF radio crash and reboot when I try to enter DevOnly menu, and presents the display issues described above. The commits before that do not display either issue. 
My old UHF md-380 does not crash, and only has the display issue when Netmon 3 is enabled (that is, a non-issue). This is true of all tested firmwares so far. I include this detail only because it might indicate where the VHF radio is having issues.

mach327, 2016-10-09:
Aha! My VHF model, when flashed, appeared to be turning Netmon3 on by itself. Reading the release notes for https://github.com/roelandjansen/md380tools/releases/tag/e762fc9, they indicated I could turn off Netmon with a long-press of the "7" key. This fixed the display issue for me! And, get this - I can also use the DevOnly menu without it crashing now, including to turn on and off Netmon 3.
Mine and @ik2cnf's issue might just be netmon3 accidentally defaulting to on, or not initializing properly, in some models.

@ik2cnf, give that a shot. Long press "7" after boot, and then try changing the channel to make the screen redraw.

sijskes, 2016-10-10:
@mach327 @ik2cnf i might have borked the compatibility mode of the configuration loading from spi flash. sorry about that. i'm testing an improved up and downgrade compatible configuration storage.

ik2cnf, 2016-10-10:
After @mach327 advice I found I can replicate the blank screen behaviour entering NetMon 3. I can restore to normal through (dirty) menus disabling NetMon.
So I suppose NetMon feature was inadvertently activated (I never did before).
On the contrary neither long-press of  "8", "9" and "#" keys  activated NetMon features nor "7" key disabled it.

mach327, 2016-10-11:
@ik2cnf I don't know why the long press keys don't work, but I would guess they need experimental and/or debug enabled.

At least we have some idea of that particular problem, but it seems @on8vm's issue is not necessarily the same.
If there's any hints as to reproducing the old problem, I'll happily try and test with my radios.

roelandjansen, 2016-10-11:
debug needs to be on for hot key support.

ik2cnf, 2016-10-11:
True, with debug on I got hot keys working.
OT: Pls, where can I find documentation about debug, experimental and devonly features?

sijskes, 2016-10-11:
for experimental debug and devonly. "use the source luke".

ik2cnf, 2016-10-11:
:-)

sijskes, 2016-10-11:
these options are really not stable enough to jusitfy documentation and the work when the meaning of those options change.

for now:
debug: enable dmesg (for now also enable hotkeys)
netmon: allow selection of netmon screen when hotkeys out of action 
experimental: very volatile in function.

hotkeys: 7,8,9,# = normal,nm1,nm2,nm3 
6 = dump syslog to dmesg
5 = clear syslog

sijskes, 2016-11-01:
closing no solution, no progress

closed 2016-11-01 by sijskes