travisgoodspeed/md380tools #426: merge mach327 commits

sijskes, 2016-10-03:
@mach327 i see your branch isn't merged yet. is the code in your last commits already in travis/master? i believe there is good info in your commits.

mach327, 2016-10-03:
I pushed some chirp driver mem_format changes back here, since I could be reasonably confident it wouldn't break anything. The other stuff, not so confident. It all lives in the master branches of my repos.

#### Chirp upload/download for travis/master:
- [ ] md380tools accessible as a module. (e.g. import `md380tools.md380_dfu` ).

#### DMR Support (overall)
- [ ] Largely dependent on Chirp getting DMR support. I have some hacky stuff that works, but I'm not happy with it - wrote much of it late at night. It does do some neat auto-programming though. 

  - [x] Non-serial based interface to radio - currently leaves that issue entirely up to radio driver when driver says to. 
  - [x] `DMRMemory` support, and teaching `Memory` class and such to be able to recognize and read DMR. Also some fixes to `Memory`'s csv parsing.
  - [ ] Graphical interface support (`DMRMemoryEditor`, etc).
  - [ ] Teach Chirp about DMR specific things (Contacts, RXGroups, and keeping references sane between them and memories). My versions, I do not like.
  - [ ] Somehow push large, scary patches back to Chirp once it seems up to Chirp quality.

If anyone, ever, looks at any of my code, and feels like using or merging it now rather than later - please do so. I don't have a lot of spare time to dedicate to it at the moment, since the day job is ramping up some projects for a deadline.

mach327, 2016-10-08:
Well, something that should make it easier to deal with later - force updated my chirp_fork repo and manually applied the current patch set, after re-cloning with git-remote-hg (wonderful tool!). It's set to be compatible with hg-git, too.

mach327, 2016-10-10:
The md380 tools can now be imported with #470 
```python2
import md380tools.md380_dfu
import md380tools.md380_fw
import md380tools.md380_gfx
import md380tools.md380_tool
```

mach327, 2016-10-24:
Closing, only difference now between mine and Travis's md380tools is one file - the chirp driver. :smiley: 

As far as I know, Chirp DMR support development has stalled - I don't think anyone is excited to write the editor, or other GUI things necessary for proper Chirp support.

closed 2016-10-24 by mach327