travisgoodspeed/md380tools #428: Mic bar on received on a18d5b8-dirty

ispkorte, 2016-10-04:
if you send and mic bar is on it will place in received mode also some time and you cant see user info
is it a bug or not ?

[image]
[image]

sijskes, 2016-10-04:
Yes, i can confirm this as a bug.

ispkorte, 2016-10-05:
i flash **d9c1676-dirty** and i check bug again but nothing found . i thing it was fix , right ?

sijskes, 2016-10-05:
@ispkorte no, it is not fixed. it is timing-dependend. sometimes the erasure of the micbar just arrives too late. or the end of tx popup too early.

ispkorte, 2016-10-05:
OK thanks

commit 57ba0017 (2016-10-06): Merge pull request #444 from sijskes/p1

closes #428
closed 2016-10-06 by sijskes