travisgoodspeed/md380tools #435: something wrong with travis?

sijskes, 2016-10-05:
See: #434  clang compiler

i can see this relating to the commits.

sijskes, 2016-10-05:
$ if [ ${TRAVIS_OS_NAME:-'linux'} = 'osx' ]; then sudo ln -s /usr/bin/python /usr/bin/python2; fi
ln: /usr/bin/python2: Operation not permitted

The command "if [ ${TRAVIS_OS_NAME:-'linux'} = 'osx' ]; then sudo ln -s /usr/bin/python /usr/bin/python2; fi" failed and exited with 1 during .

Your build has been stopped.
/Users/travis/build.sh: line 109: shell_session_update: command not found

sijskes, 2016-10-05:
i merged, and now the relation with the build is gone. it was in the early setup of the container calling several 'brew'.

sijskes, 2016-10-07:
strange thing. the ln is run with sudo. maybe /usr/bin/ is read only mounted fs?

travisgoodspeed, 2016-10-07:
Perhaps travis-ci should be running `make dist` rather than `make install`?

sijskes, 2016-10-07:
@travisgoodspeed  better. but it looks likes it goes wrong in preparing the container. i will change to make ci

sijskes, 2016-10-08:
setup of clang part is broken before make ci is called.

travisgoodspeed, 2016-10-20:
Does Travis CI build in an OS X VM?  I assumed they'd use Docker with a Linux VM.

`/usr/bin` isn't writable on modern OS X, even as root, so I'd expect that line to fail.  `/usr/local/bin/` is writable and in the default `$PATH`.

d235j, 2016-10-20:
They may have upgraded to 10.12 which introduced Rootless.
The reason for that line was to work around the python2 -> python problem. Seems we need a better solution.

travisgoodspeed, 2016-10-26:
Status screen is at https://travis-ci.org/travisgoodspeed/md380tools

Self-assigning.  I'll have this passing today.

travisgoodspeed, 2016-10-26:
Hey, that worked on the first try!  Closing issue.

commit 3a632df1 (2016-10-26): Removing sudo ln from TRAVIS-CI.  Might fix #435.
closed 2016-10-26 by travisgoodspeed