travisgoodspeed/md380tools #446: MD390 / RT8 volume knob

ispkorte, 2016-10-07:
I wonder about two radio units RT8, if I be located with my RT8 the volume knob counter clockwise to 0% then I hear still talks. 

(i don't know about a bug on hardware or a fault in firmware) is there anybody to confirm it  ?

hamradioPRC, 2016-10-07:
This is a known issue with stock firmware. Tytera and clones won't do anything to fix it am afraid. Good idea for the hackers

hamradioPRC, 2016-10-07:
This will be closed when a fix is understood

m0ivq, 2016-10-07:
The volume level itself is hardware and not controlled by the firmware (unless that has changed in later models). The only connection to the micro is to measure the position of the volume control in order to display the volume setting.

eb4fbz, 2016-10-08:
There is a resistor in series with the volume pot to ensure a minimum volume with pot at 0%, it's done on purpose so you don't miss any transmission by mistake.

ispkorte, 2016-10-08:
ok ; it is a feature , not a bug 
thanks eb4fbz
(dd1go/hs0zkj)

eb4fbz, 2016-10-08:
This is the resistor, just in case someone wants to eliminate that feature. Just bridge it  or the potentiometer pin to ground.
[image]

ispkorte, 2016-10-08:
many tnx to eb4fbx

i check it

sijskes, 2016-10-08:
hardware issue. closing.

closed 2016-10-08 by sijskes