travisgoodspeed/md380tools #448: Radioddity GD-55 / MD398G

paulomartinspt, 2016-10-07:
They have a GPS and a non-GPS version. It seems like a MD390GPS with four major differences:
- man down
- 10w
- built-in recorder
- big battery

http://www.radioddity.com/us/radioddity-gd-55-uhf-waterproof-dmr-digital-radio-10w-with-2800mah-lithium-polymer-li-po-battery-1.html

Anyone tried one? Will it load the RT8/MD390 firmware?
I've read somewhere that this Radioddity branded radio is made by TYT and it's in fact a MD398GPS.
One of the proves, see tab label: http://www.radioddity.com/skin/frontend/base/default/download/Radioddity%20GD-55%20GPS%20Operating%20Instructions.pdf

So if this is, as it seems, in fact this guy with a few chances (lower battery for example)

https://tyt888.en.alibaba.com/product/60418744196-222728557/Tytera_MD_398_very_cheap_Mototrbo_DMR_radio_2_time_slots_IP67_waterproofed_GPS_optional_3200mAh_battery_capacity.html

what are the chances of using custom firmware on it?
Bigger battery, 10w, 8h of voice recording (not that useful for hams but good for testing purposes) etc makes this a hell of a radio for this price tag.

travisgoodspeed, 2016-10-07:
I have firmware for this model, but I've not yet unpacked it.

paulomartinspt, 2016-10-07:
Cool 👍 
Thanks for the fast reply @travisgoodspeed, as always.
Do you know if it's really worth buying it? What's the catch on this radio? Because I don't understand why people are buying (at least in Portugal) the RT8/MD390 with or without GPS for the same price as the RD-55/MD398G. Why isn't this model a little bit more famous that it is?
The MD380 i understand, it costs a less, has a great support team (on github) and it works very well, but the MD390? I can't get it.
I don't know anyone that has the MD398G.

travisgoodspeed, 2016-10-07:
The 398 is very new, and even the 390 has only been out for a few months. If it becomes very popular, we might add support to this project. 

Keep this issue open at least until we've investigated the 398.

paulomartinspt, 2016-10-07:
Sure, thanks again.

xSmurf, 2016-10-08:
Howdy neighbors! I gambled and bought one of these a week or so ago (still isn't shipped :-1:) I'm glad to see you have the firmware already that's a start :)

I bought it knowing it's a gamble and thinking at worst it'll be a second radio for tests and what not. But there seems to be some major bugs/mistakes in either the CPS or the Firmware (or both) which, at the moment turns the unit into pretty much a brick. With the currently published CPS it is not possible to use TS2 at all! At the very least it appears the intern they hired didn't finish his first visual basic program in time or something. And either QC didn't see it, or Radioddity released an unfinished copy or something. Until further notice this is definitely buyers beware.

As @travisgoodspeed pointed out this radio is extremely new. There are some early references a few months back, but for the most part it was first announced around late august. With some very early adopters getting theirs around the last week of September, but without a programming cable (so essentially a brick since not even the Radio ID can be programmed in).

One thing we do know for sure: It uses a Motorola M328 or M328+ (also called M5 or M7) connector instead of the Kenwood 2 pin. We don't know which of the two either.

I hope it's ok for me to use this issue/thread to keep track of more information as it comes.

htperry, 2016-10-08:
I have one of the Radioddity GD-55, since September 26, and can't use it for lack of programming cable. They've been selling these radios for weeks without offering and delivering a programming cable. I've not been able to get any email response for the last week. I think there are other issues with the design, but the biggest barrier to using the radio as intended is inability to configure.

I tried last night to disassemble the GD-55 for pictures, but it is glued together with sealant used to insure IP67. I wasn't willing at this point to force the radio apart. It seems nicely built. I hope someone builds a firmware that makes this radio work as it should. I also hope someone will publish the schematic and necessary parts list for the programming cable.

At this point, my suggestion to anyone contemplating buying a Radioddity GD-55 is "Buyer Beware". This DMR radio is not ready for prime time.

htperry, 2016-10-10:
Just got this directly from TYT as a reply to an inquiry from me through Alibaba. The TYT MD-398 supports only Time Slot 1, as I have seen in the firmware and CPS. Isn't the 2 time slot function part of Tier II support? 
-----------------
Hi HT
Hi Pery,
MD-398 support Tier II,but please note that MD-398 is ont time slot and it have only one time slot when work with DMR repeater.
MD-398 come with 1 battery,1 adaptor and 1charger,1 antenna, but not program cable and speaker.
Regards
alice
--------------------

tirititraun, 2016-10-10:
I think this is the cable and the motorola connector
http://www.radioddity.com/us/radioddity-rd-202-programming-cable-for-radioddity-gd-55-dmr-digital-radio.html

htperry, 2016-10-10:
It is and they still don't have it to ship. They are taking orders without telling buyers they aren't actually in-stock. Plus, they are selling GD-55/MD-398 radios without disclosing that it's not 100% DMR Tier II compatible due to lack of Time Slot 2 support. Radioddity isn't to be trusted. This the the same company that has been taking orders for Baofeng DM-5R DMR radios that don't support Tier II. They are now telling people to buy the Baofeng DM-5R radios in pairs for a better DMR experience.

htperry, 2016-10-11:
[image]
[image]

on4amh, 2016-10-14:
I'm also waiting for 14 days for my programcable
I can input more then 650 contact but when I save the code plug and reloaded in to my program there are now 650 contacts but only 256 contact where is the rest ???
Also just 5 characters for call sign in contact

on4amh, 2016-10-14:
It would be nice if we can use the hacked firmware from  the md380 ( works great 😜) on the GD55
Or MD398, but no programcable😡😡😡

htperry, 2016-10-14:
As of October 14, Radioddity says they've still not received their first batch of programming cables to start shipping. I got a 50% refund on my radio and promise of free programming cable shipped DHL. Two days ago Yako in customer support said he was being told they'd have cables in 1-3 days. They refuse to answer questions about the lack of Slot 2 support. I have $54.00 in this radio with hopes someone in the aftermarket will fix it. In the meantime, I have a MD-380 being delivered today.

on4amh, 2016-10-14:
I have the same experience with radioddity and Yako, over 30 mails and at least still nothing no answers on my questions
I hope when I get my programcable we have good news from Travis that firmware 380 works on 398 or GD55 ,

roelandjansen, 2016-10-14:
Could we just close this one? The radios are not supported and so far it
seems mediocre at best.

travisgoodspeed, 2016-10-14:
Let's close this after we take a quick gander at the firmware update, but not before.

-Travis

on4amh, 2016-10-14:
Ok, thank you Travis , I'm looking forward to it

travisgoodspeed, 2016-10-14:
To be clear, I'm just taking a look to understand how much work a port would involve. I don't expect that this project will perform a port unless the 398 stabilizes and becomes popular. 

One step at a time,
-Travis

IK8JHL, 2016-10-19:
I do not know if it can be important but the software of the 398 and almost equal to Wouxon D901 which uses as the famous Prolific interface, this leads me to think that it is quite different from the MD380 .In a few days I received two copies one with GPS and one without, and I will be able 'to be more' accurate.
[image]

IK8JHL, 2016-10-19:
Its real similar , i dont understand if Wouxon copy from TYT or else , but its very different from MD380/390

[image]

htperry, 2016-10-19:
Does the Wouxun D-901 support Time Slot 1 and Time Slot 2?

I've written TYT and Radioddity about the lack of Time Slot 2 support. They both don't care as along as they sell radios. They say buy a MD-380/390 if I want TS2 support. I have a MD-380 ... and a Radioddity GD-55/MD-398 w/o programming cable. I have tracking information that says I have a GD-55/MD-398 programming cable in transit.

IK8JHL, 2016-10-19:
D901 have SL1   , SL2 , and work fine , its compatible 100% on DMR ,

on4amh, 2016-10-20:
Where can i download firmware D901, i want to try THE firmware from D901 to install on my MD398

IK8JHL, 2016-10-20:
Maybe there aren't  Hacked firmware for D901.
I say that its similar but dont equal
I think thats  MD398 dont have hardware  for trasmitting impulse TX RX data tier II

on4amh, 2016-10-21:
The KG D2000 has also gps , but I can't find the program software

htperry, 2016-10-23:
Dear htperry,

dsalomon has just replied to a thread you have subscribed to entitled - I have been seeing this Tytera MD-398 radio. - in the Budget and Entry Level Transceivers forum of The RadioReference.com Forums.

This thread is located at:
http://forums.radioreference.com/budget-entry-level-transceivers/336933-i-have-been-seeing-tytera-md-398-radio-new-post.html

Here is the message that has just been posted:
***************
I have been in communication with both the manufacturer, Radioddity, and the main reseller, Sanatop (via Amazon).  Radioddity confirms that this is a single Timeslot radio and that they have no intention of changing that, near term or long term.

A few days ago, I learned that this is NOT something that can be corrected via firmware.  I'll have to go find the posts and forum where the information was conveyed.  However, the bottom line is that there is a hardware component in the radio that controls the use of the bandwidth.  The one in the MD-398 / GD-55 is a single Timeslot only.  To change that, the component would have to be replaced and the hardware redesigned.
***************

moiei, 2016-10-24:
I have been told that the option "slot2" is disabled (commented) in the .ini file of the codeplug editor. Can any of the owners confirm if this stupid change is an actual change or if it is just a cosmetic thing?
[image]

km4vig, 2016-10-25:
I have just received my programming cable today for my GD-55. Most if not all of this is VERY new to me.

  and of course the "Slot-2" is missing from the software....

   I found an XML  file to edit .. "Comb_en-US.xml"

   I added the additional "Slot-2" and it seems to be working!

   I'm testing with my DV4Mini;

     I will try a few radio checks to verify it works on 3100 US wide.

on4amh, 2016-10-25:
Where do i edit THE xml file?
How does iT work to edit ?

on4amh, 2016-10-25:
Can you send me the modified software,or a link where I can download, many thanks

IK8JHL, 2016-10-25:
I have doing  more test , 
1)The radio transmit  on DMR only on one slot
2)Its write Slot1 but really use  Slot2
3)I have enable slot2 in the software but  dont change nothing
4I spoke with TYT and they told me that  can not add the second slot

boatman1069, 2016-10-25:
Wow i hope you guys can do something for us with a one eared dg-55.
Please please travis sir ..mr nice man sir sir really nice man ...ill buy you a coffee...hehe😜

Thankyou

km4vig, 2016-10-25:
The file you need to edit is ....  located in either 
 (x64 bit computer) **C:\Program Files (x86)\GD-55\Comb_en-US.xml**
 (x32 bit computer) **C:\Program Files\GD-55\Comb_en-US.xml**

[image]

( _I recommend using Notepad++ to edit the file but Notepad might work._ )

the first thing you need to do is copy that file to a different location because the file system will not let you change the file where it is.

I made a new folder on my C: drive called C:\tmp

 I copied it there and made my changes then copied it back to 
C:\Program Files (x86)\GD-55\Comb_en-US.xml
after my modifications.

 there is line that has been commented out: (line# 171)

  you have to remove the comment tags: "<**!--**"  &  "**-->**"
so the line looks like this:
<item key="1" value="Slot 2"></item>

Then just save the file and copy it back to it's original location ...  when prompted to "Copy and Replace"

boatman1069, 2016-10-25:
Wow if this works ok your my hero  ..im just waiting for the programming cable to turn up...i just found out the charger base is exactly the same as the md-390...

boatman1069, 2016-10-25:
So km4vig is this working now on both time slots.. please share your experience of the radio.

Thanks

on4amh, 2016-10-25:
In the link dropbox:  that is not the right file in think

km4vig, 2016-10-25:
Well Yes, I assume it's still working on Slot1 -- I have no way to test that. However I have had several QSOs on Slot2.

I am presently monitoring 3100 US NationWide .. going out to early VOTE this morning but when I get back I'll be online for the morning....

km4vig, 2016-10-25:
on4amh  - not sure what you mean ... ??

   can you explain why you think that?  you do need to "RIGHT-CLICK" and Save As ... "**Comb_en-US - Copy.xml**"

on4amh, 2016-10-25:
I saved the file to my USB stick ,  I just modify line 171, then I want to save it back to my c drive , but I get an warning when I want to save it, " entery denied "

on4amh, 2016-10-25:
Ok no I fiend the Wright way to copy paste , and I see no "" slot 1 and slot 2""
I hope to receive my programcable next week, it's on the way ...I keep you informed 
Many thanks for help. 73

km4vig, 2016-10-25:
on4amh

    glad you were able to get it working ...

  I was able to setup another "ZONE" for my local repeater - and confirmed that it was working with a short QSO --  worked very well!

IK8JHL, 2016-10-25:
Its not necessary to modify software , because the firmware and hardware dont have two slot .....you can rceive slot 1 and 2 automatically , but when transmit , the radio transmit only on Slot2 .
Attention because if use the radio for example on World , or Europe  thats normally use slot1 you move All traffic of local repeater on slot2 and other Ham tuned on local Repeater miss a comunicatio

km4vig, 2016-10-25:
IK8JHL

    tried to understand what you were saying, and couldn't ....  the part where you say that the firmware and hardware don't have slot 2 - doesn't make sense to me... as All I did was modify the config file that the editor uses to provide this optional value of "1" for the time slot field. Before it only provided the option of "0" for time slot 1. Now that that is available the codeplug is written to that field.  And the fact that it now works with time slot 2 - would confirm to me that the firmware & hardware does in fact support "slot 2"

IK8JHL, 2016-10-25:
You have enable SOFTWARE of Codeplug Editor not Firmware in the Radio, check in the menu' of Radio if you can change slot2......
You can do qso because the radio receive Slot1 and Slot2 automatically but when  trasmit you use only Slot2 .You have try to speak with another local Ham on the same Repeater?????

on4amh, 2016-10-25:
Thanks i Will try iT, I keep you informed 

73' ON4AMH
Francis

km4vig, 2016-10-25:
IK8JHL

  there is NOT a "Slot 2" in the menu of the radio.
  I'm still unsure of what you are trying to say?
  I made the change in the codeplug via the modifications to the software/editor config file, and now it works. Before that it didn't. If you are saying that what I didn't change anything; and that it's really transmitting on both Slot 1&2 that's why it's working???

  I have NOT made a QSO with anyone on the same repeater (local W4AC) with me. However I have had a QSO with someone on US Nation Wide (3100) from my local repeater (W4AC). Most of my testing however has been trough my DV4Mini - and has been working perfectly.

on4amh, 2016-10-25:
volgende geschreven:

IK8JHL, 2016-10-25:
Im sure my  TYT DM-398 transmit only on SLOT2  (but  i repeat in the menu' of Radio its write SLOT1 .) and automatically receives both the slot1 and the slot2
Make this test, leave a tuned another radio like MD380 radio on a QSO thats  are using a Slot 1, then transmit  with the DM398 and see if the DM380 continues to receive on slot1 or you mast retune  on slot2.
With MD398 you do not notice anything because as I said automatically receives both the slot1 that slot2
then if yours is different from mine (but I do not think) I'm happy for you

km4vig, 2016-10-25:
I don't use slot1 for anything so I can't test that. 

  It works for me fine as is. 

Thanks.. 73's

xSmurf, 2016-10-26:
If only someone who has one at hand would take the plunge an open up that puppy, we would have a much better understanding of it's inner working by seeing it's entrails.

sijskes, 2016-10-26:
shall we do the rest in the googlegroup? can somebody create a document that we can put up, condensing this whole thead in a few simple words?

travisgoodspeed, 2016-11-28:
Closing this issue.  This project will not be supporting the GD-55 or MD398.

closed 2016-11-28 by travisgoodspeed