travisgoodspeed/md380tools #453: firmware makefile problem

sijskes, 2016-10-07:
with make 3.61 it doesnt work conform makefile semantics.

when a target is made, and previous only a clean has been run, the wrong intermediate target is selected. 

workaround, always do a download in a separate make before calling for a real target.

test_ok.sh
test_fail.sh 
are included, and verified in the ci build containers

ispkorte, 2016-10-08:
i get error ,too

60df702..91b3155
error: Your local changes to the following files would be overwritten by merge:
        patches/2.032/Makefile
Please, commit your changes or stash them before you can merge.
Aborting
(on Mint17.3)

phl0, 2016-10-08:
You made local changes to the Makefile. That prevents an update. You may want to revert the changes before updating / merging.

ispkorte, 2016-10-08:
i have definitely nothing change ; yesterday it works fine , shutdown and today restart , pull take a fault 

my last working get 60df702

phl0, 2016-10-08:
What is the output of "git diff"?

sijskes, 2016-10-08:
please open a new issue, this is completely unrelated to the issue at hand.

travisgoodspeed, 2016-12-23:
Is this still a problem?  I've checked all of my servers, but even that OpenBSD machine that I physically lost but can reach by the network is running gmake 3.81.  (Or perhaps I misunderstand how to reproduce the issue.  Maybe a shell log would help?)

sijskes, 2016-12-23:
@travisgoodspeed problem is still there but worked around with dual step makefile trickery. closing.

closed 2016-12-23 by sijskes