travisgoodspeed/md380tools #471: Linux USB Connect for Software 1.3x

ispkorte, 2016-10-09:
hi ,
is it possible to connect original Software on Linux and wine ?
some _OldMen_ do not understand console and md380-toold 
if not , please close it 

USB is found in syslog
```
 kernel: [12635.268213] usb 3-1: new full-speed USB device number 14 using xhci_hcd
 kernel: [12635.402196] usb 3-1: New USB device found, idVendor=0483, idProduct=df11
 kernel: [12635.402203] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
 kernel: [12635.402207] usb 3-1: Product: Digital Radio in USB mode
 kernel: [12635.402209] usb 3-1: Manufacturer: @00000008 : ffffffff
 kernel: [12635.402212] usb 3-1: SerialNumber: 00000000010C
 mtp-probe: checking bus 3, device 14: "/sys/devices/pci0000:00/0000:00:14.0/usb3/3-1"
 mtp-probe: bus: 3, device: 14 was not an MTP device

```

mach327, 2016-10-09:
Using the Tytera CPS in Wine? 
From #241, I believe the upload and download to radio do not work in Wine, but I have not tried it myself. Maybe you could try? If you try it, do please give us a report of how it went.

Closing for now.

closed 2016-10-09 by mach327

xSmurf, 2016-10-23:
FWIW @ispkorte, Wine does not understand USB so don't waste your time, this can never work.

ispkorte, 2016-10-23:
thanks @xSmurf 
 it is not real right about USB and WINE but it is not needed to fix something
i have install MD380tools and have made a bash script on desktop with FAQ service

-- cloese --