travisgoodspeed/md380tools #477: Backlight control

szychor, 2016-10-09:
Is there a way to change that when radio opens squelch  it wont light up LCD ? Its very annoying when im listening it at night  it lights up everytime it  recive call and its too bright .It should light up only when i press keys. Its not very important but if its possible it would be nice to have that option.

ispkorte, 2016-10-10:
sounds as be a good idea

mach327, 2016-10-10:
I've added this to the master wish list. 
This can still be discussed here, but closing until there's some work on it. 

I like this idea too, for the same reasons.

closed 2016-10-10 by mach327

kd8uxc, 2016-12-03:
CROSS POST from: #370 due to the similarity of the request.
Perhaps my request can be combined in the event this request is taken on. My slightly different way about addressing the issue may even satisfy the needs of the OP.

What about a way to turn the light off completely. Then have it only turn on when either the menu button or the back button is pressed. When I am engaged in a QSO or listening I have no need for the light.  As the OP stated it can be very annoying in low light conditions. It's more problematic (even dangerous) in a vehicle at night. I have found myself covering the screen so I don't distract the driver.

travisgoodspeed, 2017-02-11:
Properly patched by @Wolf-DL4YHF in #651.

Wolf-DL4YHF, 2017-02-12:
Thanks Travis - I have deleted my temporary branch now.