travisgoodspeed/md380tools #483: User registered 7 days ago missing from users.csv

paulomartinspt, 2016-10-10:
File loaded last night, unknows ID
But: http://ham-digital.org/dmr-userreg.php?usrid=2681047

[image]

travisgoodspeed, 2016-10-10:
Querying the database directly, I don't find an entry for 2681047 or CT1HFS in the DMR-MARC database export at http://www.dmr-marc.net/cgi-bin/trbo-database/datadump.cgi .  At the same time, I can verify that it is shown on ham-digital.

Perhaps the dmr-marc database has been moved to a new location?

Keep this issue open until we discover why updates are being omitted or delayed.

mach327, 2016-10-11:
I asked for another ID a day or two ago, and my father just got his first, both assigned late yesterday - and we show up in the dmr-marc.net database. We also show up in the ham-digital.org database. We are both northeast USA hams.

Thoughts:

My ID showing up in DMR-MARC would seem to indicate the datadump is still showing recent data.

Are we certain that the user CT1HFS registered with DMR-MARC? 
If the user did, does DMR-MARC have different internal databases they must synchronize, or different procedures for different regions or countries?

mach327, 2016-10-11:
Reading my ID approval email, I spotted this: 
> It will work immediately on the DMR network.  Allow several days for the CCS7 database to sync up and display your callsign. 

This (CT1HFS ID) may be a case of just waiting for the callsign to propagate from one database to another.

travisgoodspeed, 2016-10-11:
Perhaps.

Does the original database entry for CT1HFS contain any accent marks?  Perhaps that's to blame.

mach327, 2016-10-11:
Nothing that prints as such on my screen. And the user did register 7 days ago - so the timing is getting fairly long for "several days".

ispkorte, 2016-10-11:
i can confirm also a number from 2623899 not include

paulomartinspt, 2016-10-11:
It is strange, that's why i report it.
I've read that it can take up to 30 days to sync but it never took this long, 7 days is a bit too long for what we're used to.

travisgoodspeed, 2016-10-11:
Does anyone have a contact at DMR MARC that we might call?  Thus seems to be there bug, but it ought to be fixed all the same.

paulomartinspt, 2016-10-11:
Not me, sorry :(
@travisgoodspeed, today it already appears.

http://dmr-marc.net/cgi-bin/trbo-database/
Users Search: 1 Results
DMR ID	Callsign	Name	Nickname	City	State/Prov	Country	Remarks
2681047	CT1HFS	Rui Alves	Ruca	Espinho	Porto	Portugal	

Nevertheless, eight days to sync a user? It could be a bug.

phl0, 2016-10-11:
Not a direct contact but I know someone who has (or should have). Will
see what I can do.

iball, 2016-10-16:
Don't know if this is related or what, but today I'm noticing more and more users showing up as unknown in the users.csv file even though they're in the users.csv file itself, with nothing showing that their entries are any different from anyone else's.
I've run make clean in the db folder and re-ran the make on it to pull down a fresh copy and loaded it on my 380 and 390 and still seeing the same thing.  DMR ID 3120096 does this now, even though he's in the users.csv file.

travisgoodspeed, 2016-10-18:
Closing this issue, as database updates are DMR-MARC's responsibility and not ours.

closed 2016-10-18 by travisgoodspeed