travisgoodspeed/md380tools #484: fix thumb function pointers

sijskes, 2016-10-11:
fix it so that in menu.c MKTHUMB i.e. (f+1) is not needed anymore.

The or of the lower bit is to signal thumb code. Ideally the compiler should generate function pointers with the lowest bit set. Because it is all thumb code.

sijskes, 2016-10-11:
To consider: https://android.googlesource.com/toolchain/gcc/+/donut/gcc-4.2.1/gcc/config/arm/README-interworking

Cannot find newer version for: "gcc version 4.9.3 20150529 (prerelease) (15:4.9.3+svn231177-1)"

sijskes, 2016-10-11:
ideally the linker should emit '|1' symbols for thumb code (all code in cortex-M series is thumb)

travisgoodspeed, 2016-10-17:
Ugh, it looks like some symbols were defined to even addresses.  To drop MKTHUMB, we'll need to convert every function symbol to odd addresses manually.  Data symbols must remain even, of course, so there's plenty of potential for breaking the build.

Close this issue when:
- [ ] We've normalized all the `symbols_*` files to odd function addresses.
- [ ] We've verified that all data addresses remain even.
- [ ] MKTHUMB usage is removed from the menu functions.

Be careful to test on all three variants when committing fixes to this issue.

travisgoodspeed, 2017-01-07:
Self assigning issue.  I'll have it cleaned up today, with all function pointers odd and no `MKTHUMB` macro.

travisgoodspeed, 2017-01-07:
As a temporary measure, the `MKTHUMB` macro is now printing warnings when the pointer is even.  We'll remove the macro entirely after all menus and all ports have been confirmed clean.