travisgoodspeed/md380tools #488: 16 digit Contact Name display garbled on TX

jruys, 2016-10-14:
When a Private Call Contact Name has maximum length (16 digits), bottom line of TX display is garbled: it shows a mix of the Contact and Zone.

Display is fine with shorter Contact Names and with Group Calls.

Can provide pictures if required.

roelandjansen, 2016-10-14:
please do. It's unclear if you are referring to the normal userid or with the users.csv option.
Also, what version do you use.

jruys, 2016-10-14:
Version is 3ffa38f (I cloned it 14/10/2016) - and referring to normal userid via the code plug.

This is transmitting with contact name "5000 Status"", all is well:

[image]

This is transmitting with contact name "Echo 9990 123456", note how the "9990"disappeared, and "MMDVM" appeared at the end (the zone is called "MMDVM"):

[image]

hamradioPRC, 2016-10-16:
MMDVM= 'Multimode Digital Voice Media'

travisgoodspeed, 2016-10-16:
This is a string termination bug in the original firmware. I'm tempted to close as "won't fix."

jruys, 2016-10-16:
Could have sworn it wasn't there in the early days of MD380 patched firmware but no problem to close - I've changed codeplug to 15 characters maximum now.

travisgoodspeed, 2016-10-17:
Closing as "won't fix."  If you encounter this bug, please do as @jruys suggests and workaround by limiting your contact names to 15 characters.

closed 2016-10-17 by travisgoodspeed