travisgoodspeed/md380tools #489: readme DE from Kim

ispkorte, 2016-10-15:
dear Kim

i m a linux user , but a windows user say to me your step by step readme was fail . so i was make a WINBOX and check step by step and found a mistaken ( i thing)
path for MAKE is lost a **;** in line
`Folgende Pfade hinzufügen: C:\Program Files\GnuWin32\bin C:\Python27`
it must be 
`Folgende Pfade hinzufügen: C:\Program Files\GnuWin32\bin**;** C:\Python27`

i hope so it will fix the problem to find right  "Path"  for _make_ and _python_

ispkorte, 2016-10-15:
a second remark for user do not work whit XP 
Start - Systemsteuerung - SUCHE "Umgebungsvariablen" to add path

travisgoodspeed, 2016-10-15:
I patched the path in my last commit, but I don't know enough to rewrite the instructions for the second part correctly.  Any volunteers, so we can close this issue?

ispkorte, 2016-10-16:
Up to date 3:)

closed 2016-10-16 by ispkorte