travisgoodspeed/md380tools #490: Firmware where only the user.csv can be used?

PD1LOI, 2016-10-15:
No error message but a request.
Is it possible to create a firmware where only the user.csv can be used, and further just standard just as the original?
At the request of a few amateurs.
greetings PD1LOI

travisgoodspeed, 2016-10-15:
Sorry PD1LOI, but we are trying to reduce the number of different release flavors that we make, rather than to make more of them.

Can you explain why they want a release with fewer features?  Perhaps we could do something with the configuration to stop whatever problem they are having.

PD1LOI, 2016-10-15:
Older amateurs find the option user.csv enough. they use the other options.

roelandjansen, 2016-10-15:
then just don't use the other options. simple as that. I also feel that too many diversions just causes headaches with the issue tracker here.

travisgoodspeed, 2016-10-15:
Closing this issue.  Future feature requests should begin on Issue #370 rather than starting as their own issue in the tracker.

closed 2016-10-15 by travisgoodspeed