travisgoodspeed/md380tools #494: Bringing back the boot chime...

xSmurf, 2016-10-18:
In stock s13.012 the boot chime plays whether the boot up message is in text or picture mode. In s13.020 however, it only plays with it is set to picture mode.

I don't know if this is something that in the realm of the possible, and definitely one that's more for the wishlist than anything. But since at least one person asked me and I noticed when the change happened, in the least it'll be documented here for future reference.

travisgoodspeed, 2016-10-18:
Moving this to #370 rather than keeping it as a separate issue.

closed 2016-10-18 by travisgoodspeed

xSmurf, 2016-10-18:
Thanks @travisgoodspeed, that's where it belong indeed.

For further documentation: it appears the boot chime works in text mode in 13.20 if the demo is disabled.