travisgoodspeed/md380tools #495: what are do the expmerintal menu and debug?

IK8JHL, 2016-10-18:
what are do  the expmerintal menu and debug?

hamradioPRC, 2016-10-18:
DON'T enable debug menu as it drains battery power.

IK8JHL, 2016-10-18:
ok thanks but what do ???? i dont find any information on the web , some people know where i can found more ?

hamradioPRC, 2016-10-18:
From the first stable release build you would enable both promiscuous and experiment modes to have last heard talk group display on screen (bottom left corner!)

sijskes, 2016-10-18:
https://github.com/travisgoodspeed/md380tools/wiki/Menu-options started

sijskes, 2016-10-18:
done. ask additional question as input for wiki.

sijskes, 2016-10-18:
leaving open, to collect response. 

before you ask extra question, please check current status of https://github.com/travisgoodspeed/md380tools/wiki/Menu-options

IK8JHL, 2016-10-18:
Ok many thanks ,now its more clear,
Another question on the Netmon , i Enable menu' " Dev Only!!! " NeMon1" 2 end 3 buto i dont understand how works the Hot Keys , i have try but dont change nothings

sijskes, 2016-10-18:
you have to have a little patience here. the document and latest release are not yet in sync.

sijskes, 2016-10-20:
docs should be in sync with head.

closed 2016-10-20 by sijskes