travisgoodspeed/md380tools #514: add bootscreen text selectable by menu option

sijskes, 2016-10-22:
Add show text from codeplug vs show text from config selectable in menu.

See: #509

sijskes, 2016-10-22:
@xSmurf are you taking this one? please assign this issue if so.

xSmurf, 2016-10-22:
Copy from the thread in the PR:

---------

@m0ivq this change displays the radio ID and your callsign, that's kind of the whole point ;) And will remind you to change your ID with a big "UNKNOWNID" for something like 0.

With that said, yeah we can potentially make it optional... I'm not sure it's worth a new menu entry though. I'd have to think about this a bit to see what is the best method here.

sijskes, 2016-10-22:
I think it should be definitely optional.

xSmurf, 2016-10-22:
But the question is under what mechanism. It appears to me that the case where usersdb is unavailable is probably pretty small considering I think it's the first and foremost feature people flash the firmware for. The feature should almost certainly be enabled by default as a major point is facilitating the onboarding process (where someone buys a radio, has an OM friend flash it for them with the firmware, the usersdb, and the local codeplug - and then they just enter their ID on radio). So what remains is what mechanism should be used to optionally disable the feature. It could be a menu, but the list is growing, and it seems somewhat overkill to me. Another option would be to detect if the top/bot lines are actually empty and only applying the mod then. But that could be annoying if one expects their ID/Name to be automatically taken care of after flashing a random codeplug.

m0ivq, 2016-10-22:
I think the issue here is, if you add an extra feature that isn't fixing something, it needs to be optional to the end user as not everyone is going to want to use it. I have no problem with the feature and think it's a good idea, just I would like the option to disable it without editing files.

xSmurf, 2016-10-22:
@m0ivq as I said, I'm not against an option to disable the feature, I'm just looking for the best way of going about that without breaking functionality.

sijskes, 2016-10-22:
started with #517 restoring old situation, but keeping all the code. an extra menu should be easy to create.

m0ivq, 2016-10-22:
Maybe add a file called boottext.txt that if it exists, it uses the contents of that, if not then uses the automatic lookup? Add the file to .gitignore so doesn't get pushed to the repo.

xSmurf, 2016-10-22:
@m0ivq that doesn't seem right... a lot of people get prebuilt firmwares and flash it with the original update utility.

xSmurf, 2016-10-22:
@m0ivq You say the feature isn't fixing anything, but this is not quite true. You seem to see only the boot splash, but this feature is coupled with automatic configuration of the radio name from the usersdb. This makes it possible for users to program codeplugs without editing them at all and getting the proper callsign setup as the radio name, and have the boot splash display relevant information (ie: call sign and dmrid - rather than callsign and name as you prefer). Not needing access to a CPS - or needing to modify codeplugs - is a pretty big thing, specially for Linux users.

Seems to me a bigger issue is that no one is writing a small UI tool for windows to take care of firmware upgrades, usersdb, and writing codeplugs ;)

m0ivq, 2016-10-22:
@xSmurf I do like the enhancement, especially as if I use someone elses codeplug the "0" will remind me to update the ID, it is only the bootsplash message not being configurable that I wasn't so keen on.

Would it be worth changing the displayed zero to a message like "Confirm DMR ID" for those (like me) who flashed it without reading the commit comments and wondered what the zero meant? (will stop others opening issues on here when they flash a prebuilt binary without reading release notes).

I am currently working on the userdb issue under windows - whether it ends up being a new UI using the standard windows driver or working out the issue with md380tools under windows remains to be seen.

xSmurf, 2016-10-22:
I'm working on a submenu for boot screens, you'll be able to enable disable the demo (currently there but to be moved to the submenu) and to configure the boot splash with three options: manual, callsign + id, callsign + name.

For now I'm considering the fact that the radio name is changed a separate issue. I will however disable that if usersdb is disabled.

m0ivq, 2016-10-22:
"Confirm" on top line in place of callsign
"DMR ID" on bottom line in place of ID?

(I haven't looked at your code so don't know how you implement the displayed zero).

xSmurf, 2016-10-22:
That might be annoying to people who willingly have a DMR ID that isn't in the usersdb and still want it displayed on boot.

sijskes, 2016-10-22:
@xSmurf why not use the text editing menu support for entering any text?

xSmurf, 2016-10-22:
seems way overkill and then users need to set the same information (in 99% of cases) twice...

sijskes, 2016-10-22:
@xSmurf this 99% is a number you just invented, right? 
Text entry gives everybody the freedom to do what she/he wants.

m0ivq, 2016-10-22:
I meant display the message in place of the 0 that gets displayed when you flash your code for the first time and (I'm guessing) if the codeplug dmr id doesn't match the one you've stored separately to check if a new codeplug doesn't match.

ID not in database shows UNKNOWNID which makes it obvious to the user so not a problem.

xSmurf, 2016-10-22:
Yes it is, but I've never seen anyone put or recommend anything else than dmrid/callsign/name in there.

If people want said freedom, they can already enter it in the CPS.

xSmurf, 2016-10-22:
Closed in #520.

travisgoodspeed, 2016-10-24:
There seems to be a bug with the LLID is zero until it's overwritten by the MD380Tools menu item.  Close this issue when the bug is fixed.  (Reproducing it might require a virgin SPIFlash image.)

Do we store the ID in two different locations?

xSmurf, 2016-10-24:
Is there a place I can get a virgin flash for the S13.20 somewhere?

> Do we store the ID in two different locations?

It's stored in the codeplug flash and global_addl_config (and of course in ram in md380_radio_config), and restored from the addl_config on boot over the codeplug flash. I've copied the same mechanism for the radio name.

travisgoodspeed, 2016-10-25:
Then the bug is certainly that the codeplug flash and global_addl_config aren't synchronized.  We should only rely on the operating system's copy in the codeplug, without storing a second copy in the addl_config.

I'll post a codeplug from a virgin radio tomorrow morning.

sijskes, 2016-10-25:
the dmrid is in the addl_config in order to override the dmrid in the codeplug.

use case: regularly distributed codeplug, you need to edit dmrid in codeplug editor every time when updating, or in the menu.

maybe the logic is incorrect, but the function a such is helpful, it allows you to flash the codeplug without having to think about updating your id.

sijskes, 2016-10-25:
proposal for the bootlines:

menu:
- reset to codeplug (i.e. no override)
- edit (i.e. editline any text, store in override)
- copy from dmr-id (copy dmrid to override without edit)

xSmurf, 2016-10-25:
@sijskes can we take the dmrid/name to another thread? this is strictly for the bootsplash. Which only need that bug fix. (Which is probably fixed in the waiting PR, but I wanna test on a virgin flash).

travisgoodspeed, 2016-10-25:
Virgin codeplugs are available in `codeplugs/` for multiple radio models.

Forked the DMR LLID issue to #535.

sijskes, 2016-10-25:
On 25-10-16 06:16, Travis Goodspeed wrote:
> Then the bug is certainly that the codeplug flash and global_addl_config
> aren't synchronized. We should only rely on the operating system's copy
> in the codeplug, without storing a second copy in the addl_config.

the dmrid is in the addl_config in order to override the dmrid in the 
codeplug.

use case: regularly distributed codeplug, you need to edit dmrid in 
codeplug editor every time when updating, or in the menu.

maybe the logic is incorrect, but the function a such is helpful, it 
allows you to flash the codeplug without having to think about updating 
your id.

G. Simon

sijskes, 2016-10-27:
see #558

closed 2016-10-27 by sijskes