travisgoodspeed/md380tools #521: Backlight timeout too long in newest firmware

szychor, 2016-10-23:
I noticed that in newest update  i can only set 30 or 60 second timeout for backlihght , cant set lower time. This is too long because after every squelch open it will light up for 30s or 1min . This will be draining battery faster . Only way to set lower setting is programm it from CPS. Can you fix that

roelandjansen, 2016-10-23:
I just checked that when I have put it to 30 or 60 seconds, it does what it says
Now, if I get back to the normal timeout settings in the MD380 menu, I can switch to 5 seconds.
And then it switches off after 5 seconds.

Unless I really do not understand what you are saying -- it works..

szychor, 2016-10-23:
Hmm in my radio when i chose 5 sec it didnt work. Still is using 30 sec or 60sec  from Md380 tools settings. I will try to flash radio again and check it again. Maybe its just weird bug .

roelandjansen, 2016-10-23:
maybe. This is the version I tested it on:

https://github.com/roelandjansen/md380tools/releases/tag/f43a36f

when set to 30/60 seconds I do see that the original menu is set to "always" -- did you see that too?

szychor, 2016-10-23:
Ok i reflashed radio again and now its working fine . When  i set 30/60 sec  in original menu its on 5s .

closed 2016-10-23 by sijskes