travisgoodspeed/md380tools #522: reinstate credits screen

sijskes, 2016-10-23:
It is now possible to configure handset to not show credits screen.

Please fix this, so a prominent credits screen is shown.

xSmurf, 2016-10-23:
I looked at doing this one, but I really couldn't get the menus to display some sort of information pane (like the existing My Number/Versions pane). Any ideas?

sijskes, 2016-10-23:
its your patch. i think it is impolite to remove credits without consultation.

xSmurf, 2016-10-23:
I mentioned it multiple times, asked to add this to the wish list, I requested help in the IRC channel to add a version pane. I've asked again for help to do it myself within the hour of you opening the ticket. AND it's not the default. Not to mention that removing the demo actually fixed a bug for me (no more boot chime).  I think I've done due diligence here @sijskes. I'm still willing to work on this, and am looking for pointers on getting a pane similar to the My Number/Versions working.

sijskes, 2016-10-23:
try sending beepcode 0 during bootscreen to silence beepsound.

xSmurf, 2016-10-23:
I don't want to silence it. I wanted it back! I did not work anymore.

nivex, 2016-10-29:
I second the wanting the radio to boot as fast as possible. If ~~shi~~ stuff is going down and I need to make a call, I don't want to wait for flashy bouncy things on my screen before I can mash the PTT. I realize folk want to get the credit they deserve for work done, but I think that can be done in an "About" menu in the settings and not interrupt what the radio was designed to do.

aeickho, 2016-12-12:
@sijskes is this functionality not included yet?

sijskes, 2016-12-12:
yes. menu selectable i believe.

closed 2016-12-12 by sijskes