travisgoodspeed/md380tools #535: Duplicate DMR IDs can Desynchronize

travisgoodspeed, 2016-10-25:
As reported in #514, we are storing two copies of the DMR LLID in the codeplug, first in the normal location and then again in a backup location.  These can become desynchronized, confusing the codeplug programmer, the boot screen, and other locations.

We should merge the two to only use the regular codeplug location.  Close this issue when there is only one DMR ID stored in SPI Flash.

sijskes, 2016-10-25:
will convert this into 'cp override' menu.

travisgoodspeed, 2016-10-25:
That'd also work.

sijskes, 2016-10-25:
pausing on cp override menu idea.

xSmurf, 2016-10-26:
I was thinking, if we could chain two (or more) of those confirmation dialogs, we could ask the user if they want to write the dmr id and lookup the name in usersdb, a sort of mini wizard... that would be ideal I think.

I've looked a little, but sadly didn't make it very far. Maybe @aeickho has an idea about this?

m0ivq, 2016-10-26:
I can see see the benefit of the 'extra' DMRID to check if the codeplug has changed. How about the following:

At startup, if 'extra' ID is blank (currently displays zero), copy DMR ID from codeplug.

If 'extra' and 'codeplug' ID's don't match (due to updated codeplug with incorrect ID) copy 'extra' ID to codeplug.

From that point on, codeplug ID is then used for everything, with the extra one used at startup to check for changed codeplug.

sijskes, 2016-10-26:
my original proposal is:
menu: 'cp override'
items: bootline1, bootline2, dmrid.
for every item: enable, disable, edit. 
for bootlines, extra: 'copy/lookup from dmrid'.

sijskes, 2016-10-26:
@m0ivq it worked like that (or should have worked like that) when i implemented it. it doesn't anymore?

sijskes, 2016-10-27:
see #442 #459

sijskes, 2016-10-27:
see #558

closed 2016-10-27 by sijskes