travisgoodspeed/md380tools #538: Simplex channel + UsersCSV

DrHyperKALICH, 2016-10-26:
Then transmitting in simplex channel then UsersCSV enabled.
This simplex channel is transmitting to "All call" group.

This ID - is the last id i am talking on repeater.

[image]

If i will transmit just after powering radio its shows this:

[image]

DrHyperKALICH, 2016-10-26:
Additional issue on UsersCSV:

Then on repeater channel receiving someone from cvs file with big info it cut info at the bottom:

[image]

At the finish of reception (then background become gray) it shows info without cut:

[image]

sijskes, 2016-10-26:
what version is this? looking at the sources only six lines should be displayed now.

DrHyperKALICH, 2016-10-26:
Retevis RT8 with d7fd089

sijskes, 2016-10-26:
strange. because the data file that should be in the handset, has no 'United States, CSS7' attached to that record.

sijskes, 2016-10-26:
try flashing a new users.csv conform the procedure in the README.md

sijskes, 2016-10-26:
still suspect about the firmware version, can you try flashing the firmware with a cleanly cloned repo?

DrHyperKALICH, 2016-10-26:
Ok, i will. Its interesting because then transmit is over on the gray background i see all information.
May be possible to increase six lines to eight?
And can you say what about 1 post (about simplex)?

sijskes, 2016-10-26:
@DrHyperKALICH did you modify the firmware yourselfs?

DrHyperKALICH, 2016-10-26:
No, just cloned and flashed.
I used old users.csv from my friend, that was a problem. FIXED now.
But what about first post in this thread? About display info then using simplex chanel with UsersCSV enabled?

travisgoodspeed, 2016-10-26:
The first post is a bit vague. How exactly do we reproduce your problem on our own radios?

DrHyperKALICH, 2016-10-26:
Just set up a any simplex channel - tx/rx the same, rx group none, any cc, any slot, tx to "AllCall" (default id 16777215 for all dmr radios - set contact type AllCall).
After just tx to this channel.
I can make screenshot of settings if needed tomorrow.

DO1FFE, 2016-11-30:
Using of Talkgroup 99 in simplex channels will fix the problem.

m0ivq, 2016-11-30:
As @DO1FFE states, the adopted convention for simplex channels (on Ham bands) is to have a talkgroup set (not all call) - in the UK we use TG9, others use TG99

travisgoodspeed, 2016-11-30:
(I think the Germans still use All Call on simplex, just to make my life difficult.)

The real bug here is that the All Call display overlaps our reorganized display, and we can close this issue when that is fixed.

DL2MF, 2016-12-01:
Recommend settings are also 1-1-1   CC1, TS1, TG1 for simplex in germany, advantage is that there is no need for a new TG like 99. Simply use the TG's already in your codeplug, also no trouble with WW TG usage, because simplex channels are nowhere routed or in use for offcial repeaters.

travisgoodspeed, 2017-01-07:
Confirmed that this is fixed in the latest builds of D13 and S13.  Closing issue.

closed 2017-01-07 by travisgoodspeed