travisgoodspeed/md380tools #560: travis-ci is borked

sijskes, 2016-10-28:
this time in 'C' compile stage (.1)

sijskes, 2016-10-28:
travis_fold:end:apt-get.diagnostics

The command "sudo -E apt-get -yq --no-install-suggests --no-install-recommends --force-yes install gcc-arm-none-eabi libstdc++-arm-none-eabi-newlib" failed and exited with 100 during .

Your build has been stopped.

sijskes, 2016-10-28:
cascading error from first:

Unpacking replacement locales ...
dpkg: error processing /var/cache/apt/archives/locales_2.24-5_all.deb (--unpack):
 trying to overwrite '/usr/sbin/update-locale', which is also in package libc-bin 2.15-0ubuntu10.15
dpkg-deb (subprocess): subprocess data was killed by signal (Broken pipe)
dpkg-deb: error: subprocess <decompress> returned error exit status 2

sijskes, 2016-10-28:
lets wait this one out. probably packaging error. (strange doesnt occur often in ubuntu).

sijskes, 2016-10-30:
there is some distro mixing going on in travis-ci. debian with ubuntu. clearly not working right now. shall we build on ubuntu as a replacement?

travisgoodspeed, 2016-10-30:
Whatever fixes the issue. We could cat /etc/apt/sources to see exactly what sort of distro they are suing.

sijskes, 2016-10-30:
works again.

sijskes, 2016-10-30:
commit: https://github.com/travisgoodspeed/md380tools/commit/777527141d8307fc889dc963e7adf211260d49ad

closed 2016-10-30 by sijskes