travisgoodspeed/md380tools #561: flashing firmware/bin/D013.020.bin causes bootloop

sijskes, 2016-10-28:
if flashing firmware/bin/D013.020.bin with md380-dfu, it stays in slow reboot loop.

travisgoodspeed, 2016-10-28:
Perhaps you have old-codec hardware, in which case D003.020 is the necessary firmware?

aeickho, 2016-10-28:
What Type of Radio you are flashing? 

MD3[89]0-{no gps,with gps}-{new vocoder alpu, old vocoder alpu}

Alex

sijskes, 2016-10-28:
D03 works. any pointers how to determine radiotype? sub-md380 that is?

aeickho, 2016-10-28:
Then it is a Radio with old vocoder alpu (licences chip) .. we can rewrite the licence test on [DS]13.020 .. not only disabling the check..  but no reboot if wrong or incorrect licence else flag= new vocoder.. then we can print this.

sijskes, 2016-10-28:
@aeickho no probl. it was about flashing original.

travisgoodspeed, 2016-10-28:
There's no way to tell from the Recovery bootloader without starting our patched firmware. Instead of "make flashorig" you'll need to run something like "make flashorig d003.020" to specify the version. 

Closing this issue now that the mystery is solved.

-Travis

closed 2016-10-28 by travisgoodspeed