travisgoodspeed/md380tools #567: Bizarre setting reset/change

samlachance, 2016-10-28:
Today, when I turned my radio on, I noticed that the boot screen for whatever reason changed back to the static welcome image (was previously set to display my callsign/phone number).

After further investigation, it turns out that several settings were changed including:

Key beep re-enabled 

Radio Info
 -My Number -> Name: Set to "HEHEHEHEHEHEHEHEHEHEHE"
 -Versions -> CP Ver.: Set to "V??.??"

My radio's code plug otherwise seems to be intact. Can still make calls and all of my talkgroups/zones still appear. This appears to be a completely random settings reset or alteration. I was using the radio earlier today without any issue.

travisgoodspeed, 2016-10-28:
Simon, could this be related to recent patches for override settings?  

Sam, please read your codeplug out of the radio.  Are settings also changed there, or only in the radio?

sijskes, 2016-10-28:
@travisgoodspeed looks like an out of bounds write. maybe caused by an snprintf wheres no room for terminating zero.

sijskes, 2016-10-28:
@samlachance working on it. next patch will contain config reset function. you need to do that, and redo your md380tools config.

samlachance, 2016-10-28:
@travisgoodspeed @sijskes Guys, thank you for the quick response. Keep up the great work!

sijskes, 2016-10-28:
@samlachance can you pull last version, and do a config reset?

samlachance, 2016-10-28:
Ok, just pulled the latest commit, flashed the radio and then did a config reset. Is there anything else that I should do on my end?

sijskes, 2016-10-28:
@samlachance  no, thats it. you found a severe but not often triggered error.

stay vigilant! :) and thanks for the report.

samlachance, 2016-10-28:
Sounds good. Once again, thank you guys for providing this great firmware for free. I, and many other hams, are very thankful!

73
N3SAM

closed 2016-10-28 by sijskes