travisgoodspeed/md380tools #573: Radio S-version appears dead

PD3LTX, 2016-10-29:
Flashed my md390 by command:
$ make flash_S13
Radio appears dead, screen completely black. No response. No audio.

Luckily still capable to upload a new firmware.... so I uploaded an older firmware from 2016-10-28. All is fine again.

roelandjansen, 2016-10-29:
it might be handy where you got the image from and which version (git hash)

PD3LTX, 2016-10-29:
I build it around 17:00 hours (CET) fresh from github. I always delete the whole directory and build it from scratch. Tried to flash it twice. Same result. Eventually I took a ready-build one from PD0ZRY which recovered my MD390. Hope this helps you to find the error. 73 de PD3LTX

sijskes, 2016-10-29:
You have to be more specific. We cannot do much with times. You have to report the last commit hash of the version that worked, and the version that doesnt work.

PD3LTX, 2016-10-29:
I've tried every pre baked firmware from PD0ZRY. Everyone works. So my conclusion so far is that it must be the new flash command. Until today I used $ sudo make flash_s013.020. Today I discovered that this no longer works. So I've used the command: $ sudo make flash_S13. Did I miss something here or is the mechanism wrong?

PD3LTX, 2016-10-29:
Last bit of the output:
cp dl/S013.020.bin bin/S013.020.bin
../md380-fw --unwrap bin/S013.020.bin unwrapped/S013.020.img
DEBUG: reading "bin/S013.020.bin"
WARNING: Funky header:
4f 75 74 53 65 63 75 72 69 74 79 42 69 6e 00 00
4a 53 54 35 31 00 00 ff ff ff ff ff ff ff ff ff
30 02 00 30 00 40 00 47 00 00 00 00 00 00 00 00
01 10 02 03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f
20 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
00 c0 00 08 00 40 0f 00 ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
Trying anyway.
INFO: base address 0x800c000
INFO: length 0xf4000
DEBUG: writing "unwrapped/S013.020.img"
make[4]: Map '/home/userdir/md380tools/firmware' wordt verlaten
make[3]: Map '/home/userdir/md380tools/firmware' wordt verlaten
cp ../../firmware/unwrapped/S013.020.img unwrapped.img
./patch.py
Creating patches from unwrapped.img.
Patching hword at 08034a60 to 46c0
Patching hword at 08034a62 to 46c0
Patching hword at 08034a76 to 46c0
Patching hword at 08034a78 to 46c0
Patching hword at 08034a8c to 46c0
Patching hword at 08034a8e to 46c0
Patching hword at 08034aa2 to 46c0
Patching hword at 08034aa4 to 46c0
Patching hword at 08034ab8 to 46c0
Patching hword at 08034aba to 46c0
Patching hword at 08034ace to 46c0
Patching hword at 08034ad0 to 46c0
Patching hword at 08049f9a to 46c0
Patching hword at 08049f9c to 46c0
Patching hword at 0804a820 to 46c0
Patching hword at 0804a822 to 46c0
Patching hword at 0804aae4 to 46c0
Patching hword at 0804aae6 to 46c0
Patching hword at 0804af70 to 46c0
Patching hword at 0804af72 to 46c0
Patching hword at 0804b9fc to 46c0
Patching hword at 0804b9fe to 46c0
Patching wide string at 080d0b6c to 'MD380Tools Ver.'.
Patching wide string at 080f9a94 to 'No Fix'.
Patching string at 080d0eb4 to 'Patched MD-380/390G'.
Patching range from 0809bda8 to 080d0614 to FF.
Patching word at 0800c020 to 080fa105
Patching word at 0800c004 to 0809bf01
Patching hword at 0809bf00 to 4840
Patching hword at 0809bf02 to 2100
Patching hword at 0809bf04 to 3901
Patching hword at 0809bf06 to 4508
Patching hword at 0809bf08 to d100
Patching hword at 0809bf0a to 483c
Patching hword at 0809bf0c to 4700
Patching word at 0809bffc to 080fa105
../../md380-gfx --firmware=patched.img --gfx=replacement-font-big-latin.pbm fontreplace
DEBUG: reading "patched.img"
DEBUG: writing "patched.img"
../../md380-gfx --firmware=patched.img --gfx=replacement-font-small-latin.pbm fontreplace
DEBUG: reading "patched.img"
DEBUG: writing "patched.img"
../../md380-fw --wrap patched.img patched.bin
DEBUG: reading "patched.img"
INFO: base address 0x800c000
INFO: length 0xf4000
DEBUG: writing "patched.bin"
make[2]: Map '/home/userdir/md380tools/patches/s13.020' wordt verlaten
cp ../patches/s13.020/patched.img ./experiment.img
python2 merge_s13.020.py experiment.img main.img 0x809c000
Merging an applet.
Loading symbols from main.img.sym
Merging main.img into experiment.img at 0809c000
../md380-fw --wrap experiment.img experiment.bin
DEBUG: reading "experiment.img"
INFO: base address 0x800c000
INFO: length 0xf4000
DEBUG: writing "experiment.bin"
make[1]: Map '/home/userdir/md380tools/applet' wordt verlaten
./md380-dfu upgrade applet/experiment.bin
Beginning firmware upgrade.
Writing firmware:
0% complete
11% complete
22% complete
33% complete
44% complete
55% complete
66% complete
77% complete
88% complete
100% complete, now safe to disconnect and/or reboot radio

sijskes, 2016-10-29:
Verified by PD2EMC. 

make mostlyclean flash_S13 

works with the current version.

closed 2016-10-29 by sijskes