travisgoodspeed/md380tools #580: add audio feedback for visually impaired hams

sijskes, 2016-11-02:
would it at all be possible to make the MD-380 or MD-390 accessible for visually impaired hams by:

- either using speech or CW to put out the contents of the screen?
- Including menu’s, settings, callsigns, talk-group information etc?

Andor PA9D

sijskes, 2016-11-02:
baisc morse code framework is at foundation level. rapid beeps get concatenated. load dependend.

travisgoodspeed, 2016-11-02:
The MD280 has no screen, but instead uses audio to announce the channel and zone numbers.  Its English is a bit buggy, and it cannot announce call signs, but it is quite usable without sight.

I like the Morse code option, especially if we could do a smooth sine wave as a background process to the audio decoding.

sijskes, 2016-11-02:
@travisgoodspeed MD280 same FW?

sijskes, 2016-11-02:
PA9D: minimum useful would be channel name.  when CW readout is used, 18 words/min should be target, or else it would take too long. readout channel numbers is no added value.

sijskes, 2016-11-02:
18 WPM dit time = 230 ms. there is significant jitter in the beep generation, enough to stick 2 dits together.

Wolf-DL4YHF, 2017-02-19:
I turned my MD380 into a 'morse generator' for a test (generating the Morse pattern in SysTick_Handler, timing is almost perfect at 20 WPM). It's in an early beta stage still, only sends dummy text, and even interferes with TRANSMITTING (because it doesn't care about the radio's RX/TX/Squelch state). Any ideas how/where to retrieve the currently selected menu text (preferrably as a plain old 8-bit ASCII string), to send it into the Morse generator's own transmit FIFO ?

Wolf-DL4YHF, 2017-03-09:
Despite some of the menu items not being properly reported, I uploaded everything as-is to my branch.
When navigating to the 'MD380Tools' menu item, the Morse narrator says "GPS / BEIDOU INFO" instead, etc etc.. but anyway, it properly reports channel names (or numbers, depending on the configuration).
It's going to take some time until being fit for a pull request, so if anyone here wants to help / debug the menu-related stuff: the branch is at https://github.com/Wolf-DL4YHF/md380tools  .

KB5ELV, 2017-03-24:
Thanks for tackling this. I, for one, will take any accessibility I can get and don't mind morse at all. Wish I was smart enough to help make it better, but I just got this MD380 and don't really know anything yet. I looked up the MD280 that was referenced, but it doesn't have many channels and costs about the same as the 380. Reckon I'll install this just for grins and haha's.

KB5ELV, 2017-03-25:
@wolf-DL4YHF ... I think I've got your branch of the firmware installed. Where do I find the morse narrator?

Wolf-DL4YHF, 2017-03-25:
It's all described here (on my website, in plain old HTML with a few 
screenshots):

http://www.qsl.net/dl4yhf/RT3/index.html#morse_output

Basically: "Menu".."Utilities".."MD380Tools".."Morse Output".  Set 
"Mode" to either "Short" or "Verbose".

Cheers,
   Wolf .

KB5ELV, 2017-03-25:
Howdy,

Thanks. Hopefully I can figure out how to turn this on :) 

Vy 73,

Wolf-DL4YHF, 2017-03-25:
You're welcome. Please let me know how you get along with it, or if / 
how it can be improved.
I'm not happy myself with the deeply nested menu structure; I already 
considered using an own menu (not the stuff based on Tytera's firmware) 
which could be opened with the "red key" in the upper right corner of 
the keypad.
The text in our "own" menu would be much easier to read out in Morse code...

73 + gl,
   Wolf DL4YHF .

KB5ELV, 2017-03-25:
Hi,

I'll definitely let you know how it goes. Looks like I'll need to get some working eyeballs to turn it on initially. Some way to bring up the narrator easily would be nice, but it sounds like it also might be hard to do given Tytera's menu structure. 

Anyway, I know there are other blind hams that will find this really helpful. Wish I was more conversant with the coding, but I'm not quite that smart :( 

Will keepyou posted. 

Vy 73,

KB5ELV, 2017-03-25:
Howdy Wolf,

Well, so far so good. My favorite daughter helped me turn on the morse output, and from there I was able to set it up to my preference (30 wpm, 650 hz). Got the side button programmed for morse narrator. Looks like most menus read. Also nice, a button press interrupts output. (As important as talking...popular wisdom is that when building a screen reader, the first thing you do is get it to talk. The second thing you do is to get it to shut up. Hey, that's sort of like kids, isn't it?) 

Only a couple things spring to mind for possible improvements some day, though, as you said, it's very usable now. 

* A repeat key. The narrator key, which reads zone/Channel/battery voltage at a press, is outstanding. I can really see this as giving more information than just about anything else. A "Repeat" key would be useful though. For example, say you missed what the menu you're on was, or somehow it didn't output, as sometimes happens. A key to repeat the last sent text would be useful for that. I could see, for example, programming long press for the verbose narrator and short press for repeat last. That's probably the biggest improvement at the moment. 

* I notice sometimes a change won't read back, seems to be in the menus. Going past and then back to the missed menu function seems to make it read what was missed. I think you knew about that. 

* I'll have to play with it more, but I don't think pitch and volume make much difference. Could be wrong though. 

* Believe it or not, I'd actually like one level faster, say 35 WPM or so. Yeah, I'm a freak. :) 

Thanks again, this is great stuff.

KB5ELV, 2017-03-25:
Hi,

Just quick, I lied. Volume and pitch do work correctly, I'm not sure what I was doing to think they weren't.

Wolf-DL4YHF, 2017-03-25:
Hello Buddy and all,

Thanks for testing and for the suggestions. Two "QRQ" modes are already 
added (35 and 40 WPM), and a method to repeat the previous message 
(channel or menu item). I will upload the new version in a few days. But 
I found a few other bugs myself today when testing, for example it seems 
impossible to retrieve data from the codeplug in a menu, like the 
contact list or the zone names. Does any of the experienced developers 
know why such strings don't appear as a string in the array 
md380_menu_memory , structure menu_t, "entries", "label" ? I waded 
through the disassembly looking for a buffer in which those items are 
stored, but found nothing so far.

There is also a problem when the CW announcement is interrupted by a 
received signal in DMR or FM. It only happens occasionally, but this 
kind of bug is more difficult to fix than a bug that *always* happens.

Have a nice weekend (ours is an hour shorter than usual, daylight saving 
time, so time to go zzz...)

73,  Wolf .

KB5ELV, 2017-03-26:
Howdy Wolf and others,

Yep. The intermitten bugs are the worst. I also noticed that about some things not reading. Still, it's a whole ton better than what I had before the update, which was nothing. I bet other folks would agree that other keypad beeps that aren't morse are too loud. Now with the morse readout turned on, I've turned off the keypad beeps, which works out great. 

Now for my next trick...have to get the user DB uploaded. ...

KB5ELV, 2017-03-26:
OK...another question. 

I've flashed the userdb. I think I have, anyway, no errors and was told to reboot my radio. (I did this on the Mac...after fighting with Windows and having the Linux machine in a not very accessible place, hey, nice I can program this radio from the mac.) However, sometimes I get
"Missed missed list"
sent even after the userdb was flashed and I seleced "enable" on usercsv. It appears the file is intact, as md380-tool lookup worked and showed me (I really want to change my name!) So...did I miss a step flashing the userdb, or is something just not yet implemented, thus the "missed missed list"?

Wolf-DL4YHF, 2017-03-26:
Is that "missed list" on the screen, or even audible in Morse code ?
I didn't check myself, the combination of morse encoder and user 
database lookup is still on the to-do list.

All the best,
   Wolf .

KB5ELV, 2017-03-26:
No idea if it's on the screen or not, but it sends in morse. That's probably what it is though, I'm guessing.

Wolf-DL4YHF, 2017-03-26:
Ok, rrr ... I will try what happens here (in Morse code), depending on a 
known or unknown DMR-ID being received.
Unfortunately there's no activity on any of the local repeaters, as 
usual when you need a signal for testing. And the next Motorola repeater 
(where timeslot 1 is always good for worldwide traffic) is out of reach.

cul,
  Wolf .

KB5ELV, 2017-03-27:
You can definitely get the "missed missed list" output if you connect to 4022. Not sure if I get it anywhere else or under what circumstances, but that one would definitely reproduce.

Wolf-DL4YHF, 2017-03-29:
Hello Buddy and all,

I still could not reproduce the "missed list" effect here. I also 
neither find the string "missed" nor "list" anywhere in the firmware, so 
I guess the database loaded into your radio's SPI-Flash is possibly not 
ok. Maybe you can read (electronically) what's inside the 
machine-generated "user.bin" file (sounds like binary but it's plain 
text) ? Or send it as a zipped archive in an email attachment for 
inspection ?
In the meantime I have extended the morse output a bit, and added a way 
to turn it on without having to navigate through the menu: When pressing 
the 'BACK'-button in the upper right corner of the keypad within 9 
seconds after power-on, the morse output will be activated regardless of 
the previous settings (in the menu).
Also, output speeds of 35 and 40 WPM have been added. And the CW pitch 
for a menu title is slightly different than the text of the currently 
focused menu item. It sounds a bit funny (like two CW stations talking 
to each other) first, but one gets used to it quite quickly, and it 
avoids confusion where the menu title is the same text as the first item 
in that menu (e.g. "Contacts").

Still missing:
   - output of the entries in the "Contacts" and "Zone" menu (these seem 
to be special cases)
   - proper documentation of the new features
   - testing, testing, and more testing ...

73,
   Wolf .

KB5ELV, 2017-03-29:
Hmm, odd. I think I only get it on that reflector (4022). After the "link to 4022" announcement peculiar to that reflector. 

Anyway, going to get the new stuff now. :) Thaks much.

KB5ELV, 2017-03-29:
Ummmmmmm. Nope, guess file attachments in email don't work. Take 2. 
[typescript.txt](https://github.com/travisgoodspeed/md380tools/files/880599/typescript.txt)

Wolf-DL4YHF, 2017-03-30:
Ok, my fault.
Since I don't have a radio with GPS I only do
'make clean image_D13'
but I don't compile the S13 firmware (for GPS).
There are some symbols (variables, and their addresses) unknown for the 
S13-based build, which causes the compiler to throw several errors. Will 
be fixed later - until someone can locate the addresses in the S13 
firmware, some of the Morse output features will simply not work.

Cheers,
   Wolf .

p.s. is the 'whitesticker' reflector #4022 the same, worldwide on 
DMR-plus / Hytera ?
If so, I can try to link it to the local (DMRplus-) repeater for a test.

KB5ELV, 2017-03-30:
Hi Wolf,

Yeah, that's the right reflector. 

Can I build everything with the exception of the GPS firmware? I don't have a GPS radio either.

Wolf-DL4YHF, 2017-03-30:
yes, to build only the D13.020-based experimental firmware: Enter

make clean image_D13

in your favourite command shell, with the current directory set to 
md380tools.
After that, to download the firmware into the radio:

python md380_dfu.py upgrade applet/experiment.bin

But since I almost finished the modifcations to get 'make clean dist' to 
run without compilation errors, it's easier to wait for say one or two 
hours, until the branch at github is updated.
Cheers,
   Wolf .

PD1LOI, 2017-03-30:
Where can I find this firmware also encourage you to test on my MD380?

Wolf-DL4YHF, 2017-03-30:
https://github.com/Wolf-DL4YHF/md380tools

At the moment, it's not fit to pull a merge-request for it yet. It can be compiled for S13 now, but will lack some of the features there because some addresses to retrieve displayed strings are unknown.

Wolf-DL4YHF, 2017-03-30:
I uploaded the modified branch again now. "make clean dist" now 
successfully builds all three firmware variants again, but I can only 
test the "D13.020"-based. Murphy will make sure the Morse output won't 
work on GPS radios. Anyone here who can test, and hopefully prove I'm 
wrong ?

A few notes about how to activate the Morse output without wading 
through the menu is explained on my website (link given further above) : 
Hold BACK button pressed before powering on, and release when the radio 
starts announcing the current channel name in Morse code. This switches 
back to 18 WPM, 650 Hz pitch, and beep volume controllable via pot (not 
perfect yet, but works to some extent).

Cheers,
   Wolf .

KB5ELV, 2017-03-30:
Thanks Wolf. I got it going a couple hours ago, works a treat at 40 WPM :) 

Have at least two guys on the blind-hams email reflector asking for this, so when you've updated the repository, will shoot over to them as well. :) Thanks again.

Vy 73

W6LDS, 2017-04-17:
The CW does fine. Only problem I've had with it is that it gives my received audio at least on analog a very high pitched tone.

KB5ELV, 2017-04-17:
Yeah, occasionally this happens. One thing that helps is to turn the roger beep on. For some reason that makes the confirm tone when someone unkeys a lower pitch.

Wolf-DL4YHF, 2017-04-17:
The short high-pitched pulse is just an aid for debugging.
It indicates a problem with the audio power amplifier control:
The Morse generator wants to turn it off after the end of a message, but 
for some reason it cannot (eg. receiver unsquelched because of a voice 
reception). Just leaving it on caused another problem : It would drain 
the battery, because the audio PA draws some extra milliamps even 
without a tone in the speaker.

I will remove the short "indicator beep" as soon as I have found a 
reliable way (for FM and DMR) to turn the audio power amplifier off at 
the right time.

Cheers,
   Wolf DL4YHF .

Wolf-DL4YHF, 2017-04-23:
I pulled an intitiall merge request for the Morse output / "Narrator" feature now (PR #734) .

Since the changes (compared against Travis' master repo) are huge, and there were lots of other modifications in the meantime (related with netmon, talker alias, edit talkgroup on the fly, etc), it will take some time until being merged. Because in the meantime I want to developing it further, the latest additions will always be in my temporary branch / fork already mentioned above.

73, Wolf DL4YHF .

travisgoodspeed, 2017-05-09:
Closed by @Wolf-DL4YHF in #731.

closed 2017-05-09 by travisgoodspeed