travisgoodspeed/md380tools #589: Users.csv only contains a single DMR ID for a callsign

iball, 2016-11-05:
I have 3 separate DMR IDs but the users.csv only contains one of them and not the other two.

AndyTaylorTweet, 2016-11-05:
Check that your users.csv actually updated, I've been having some issues with updating my userdb (submitted PR #590 to fix the issue I had)

iball, 2016-11-05:
Nope.  It's creating a new users.csv just fine it's just not pulling ALL of the DMR IDs for those callsigns that have multiple IDs.

AndyTaylorTweet, 2016-11-05:
Checking the users.csv I have, I can see that the duplicates *ARE* in it.
To list all duplicate entries;
in md380tools I ran:
awk -F "," '{print $2}' user.bin | sort | uniq -d

This gave me a HUGE list of users with more than one CCS ID, so I have no doubt these are making it into the radio.

iball, 2016-11-05:
I'm not seeing anything like that, I'm only seeing single entries.  
Are you sure your users.bin is newer than the last commit?

AndyTaylorTweet, 2016-11-05:
I did say my pull request #590 fixes issues with the users.csv creation....

iball, 2016-11-05:
Andy, I used your fix and it did not resolve the issue.

AndyTaylorTweet, 2016-11-05:
do you get any errors when you run "make updatedb" ?
How large is the md3680tools/db/users.csv and md380tools/db/stripped.csv

Mine are 3.1MB and 2.9MB respectively, both freshly created in the last hour while I was working on fixing the socket errors I was getting while trying to update them...

If yours are significantly smaller than this - you are clearly missing a load of data.

sijskes, 2016-11-05:
@iball could you be more specific. we need this information in order to reproduce the bug. just informing us it doesn't work helps nobody.

what id's do you use?

iball, 2016-11-05:
No errors when it's running.

I use DMR ID's 3120183, 3120065, and 3120079.  

users.csv is 2.7MB and stripped.csv is 2.5MB.

stripped.csv has 2,536,890 lines in it.

Here's the output from it:

http://pastebin.com/raw/2tmDauNU

AndyTaylorTweet, 2016-11-05:
I get all three of your callsigns in my user.bin.

While I was having problems with the generation tonight - I did wipeout my md380tools and go a fresh pull (and then fixed the issues with the download).

Could be that you are getting some broken copy of the files due to a localised caching issue, as you have realised by now, you are missing 500,000 lines from your file.

iball, 2016-11-05:
I deleted my md380tools directory and did a fresh git clone.  I even shut down the squid proxy caching server on my firewall.
Still having the same issue.

It seems to somehow be skipping anything above 3120064 for some odd reason:

3120050,KC0LDA,Douglas J Kelly,Doug,Olathe,Kansas,United States,DMR
3120051,WB0KIA,Kenneth C Yarnevich,Kenny,Kansas City,Kansas,United States,DMR
3120052,KC0DAA,David C Stearns,,Olathe,Kansas,United States,DMR
3120053,KF0XU,Lawrence D Meyer,Larry,Kansas City,Kansas,United States,DMR
3120054,KE0IRV,Dawna R Mc Gill,,Kansas City,Kansas,United States,DMR
3120055,N0KQX,Dale E Urban,,Garden City ,Kansas,United States,DMR
3120056,KE0HRB,Michael T Solomon,Todd,Plainville,Kansas,United States,DMR
3120057,AD0UN,Harald Doering-Powell,,Overland Park,Kansas,United States,DMR
3120058,KC3OL,Theodore S Holdahl,Ted,Lawrenc3,Kansas,United States,DMR
3120059,KF0RS,Daniel D Harlow,,Bucyrus,Kansas,United States,DMR
3120060,N0TTN,Brian E Williams,,Olathe,Kansas,United States,DMR
3120061,W0PRH,Paul R Heintz,Pheintz,Gardner,Kansas,United States,DMR
3120062,KE0AEP,Keith E Mckinney,,Overland Park,Kansas,United States,DMR
3120063,W0DRN,Dennis Nealey Nealey,,Prairie Village,Kansas,United States,DMR
3120064,K0GBW,Robert S Porter,,Shawnee,Kansas,United States,DMR
3120101,N0MJS,Cort Buffington,,Lawrence,Kansas,United States,Portable
3120102,KD0CYJ,Shannon O�connor,,Lawrence,Kansas,United States,Portable
3120103,N0MJS,Cort Buffington,,Lawrence,Kansas,United States,Portable
3120104,N0RC,Reid Crowe,,Lawrence,Kansas,United States,Portable
3120105,WV0T,Phil Leonard,,Overland Park,Kansas,United States,Portable
3120106,WV0T,Philip Leonard,,Overland Park,Kansas,United States,Mobile
3120107,KC0IDF,Lou Eckler,,Lawrence,Kansas,United States,Portable
3120109,WD0EMR,Ken Johnson,,Lawrence,Kansas,United States,Portable

iball, 2016-11-05:
Even when doing a raw dump into a users.json file, it's still skipping DMR IDs.

I did this:

curl -L -f 'http://www.dmr-marc.net/cgi-bin/trbo-database/datadump.cgi?table=users&format=json&header=0' > users.json

and it's still skipping:

**{"radio_id": "3120064","callsign": "K0GBW","name": "Robert S Porter","nickname": "","city": "Shawnee","state": "Kansas","country": "United States","remarks": "DMR"},{"radio_id": "3120101","callsign": "N0MJS","name": "Cort Buffington","nickname": "",
"city": "Lawrence","state": "Kansas","country": "United States","remarks": "Portable"},**

sijskes, 2016-11-05:
i thinks there is some funny going on with the networking between your VM and the DMR-MARC db.

i can fetch the file here with 3120064 in it. try again in 24hr. maybe it is temporary.

iball, 2016-11-05:
Yes, I can fetch the file with 3120064 as well, but everything from 3120065 to 3120100 is completely missing.

sijskes, 2016-11-05:
ah, not a tools problem. please ask DMR-MARC about it.

closed 2016-11-05 by sijskes