travisgoodspeed/md380tools #592: MD3x0 GPS does not work on motorola and mmdvm (only on hytera)

Giorgiofox, 2016-11-07:
The 3x0 GPS not work on MMDVM and Motorola Repeater:
it reports to send 8 blocks of data but it only sends 2

on my MMDVM for example I have:

DMR Slot 2, received RF data header from 2222280 to 5057, 0 blocks
DMR Slot 2, received RF data header from 2222280 to 5057, 0 blocks

Seem to work only on Hytera repeaters (I cannot test)

travisgoodspeed, 2016-11-12:
@Giorgiofox Do you also see this problem with the original firmware, or only with the md380tools patched firmware?

I think that the Brandmeister folks had to manually add support for GPS to their software, but I don't have the details.

sijskes, 2016-11-12:
Comments from BM dev Rudy, there is something wrong with the 'DPF' (data packets to follow) count.
I have the hardware here to debug. But i havent found the time yet.

Giorgiofox, 2016-11-13:
As I can see the problem happen also with original firmware, so is not really an issue of the patched one, but maybe we can patch also this problem ;)

on4amh, 2016-11-17:
I have a md380 with gps , can i update with md380tools so that the GPS still works?

sijskes, 2016-11-17:
@on4amh this is not a forum. this is an issues list. please ask your question at the support forum:

https://groups.google.com/forum/#!forum/md380tools

travisgoodspeed, 2017-01-21:
Unlocking this conversation so that @Giorgiofox can contribute.  I really wish there were something like IRC's concept of voice.

Is there any progress on understanding this, or should we mark it as WONTFIX and move on to other issues?

ok2it, 2017-02-01:
Hi
I have MD-390GPS, Hytera AR685(GPS) and MOTO DP3601(GPS).
It seems that both MOTO and HYT report that they are about to send "X" number of blocks UDP and send where MD390G reports X and sends Y. Could this be relatred to DATA CALL CONFIRMED and NOT CONFIRMED perhaps not working correctly?? Its just a thought....

PS -  I would love to see GPS working. I selected 390G for GPS.
PS2 - Via MMDVM / local traffic - friends with MD390G also, transmitts and my MD390G displays his GPS coordinates. So the GPS data are making it through in some form??

Jiri
73

travisgoodspeed, 2017-02-01:
Closing this issue as WONTFIX.  We'll wait and hope that Tytera handles it.

closed 2017-02-01 by travisgoodspeed

eb4fbz, 2017-04-30:
Please don't fix this while admit criteria doesn't work properly #120  
Enabling automatic position report without polite admit criteria will create interference.