travisgoodspeed/md380tools #603: Display Talkgroup 26270 as 2627..

DL9SK, 2016-11-17:
There is a problem for displaying the Talkgroup 26270. In the display it is written as 2627.. (two dots as the end instead of 0)

I tried 26271 as talkgroup and it will been shown correctly.

[image]

73, Ralf DL9SK

DL9SK, 2016-11-29:
display.c :   Removing space before and after _->_ solves the problem with long TG numbers. Then it will fit into the display line.

gfx_printf_pos( RX_POPUP_X_START, y_index, "%d -> TG %d", src, dst )

Maybe it is possible to remove this in the code.  I am not familiar with Github to to this.

gfx.h : There is a definition for length max 18 characters. Is the number limited in the stock firmware?

// max strlen = 18, if larger end in ".."

If not, where I will find the maxstrlen for expanding to 20 characters

73, Ralf DL9SK

sijskes, 2016-11-29:
It is baked in the firmware.

I've tried to enlarge the number, but it seems the interal buffer is 
limited to 19.

https://github.com/travisgoodspeed/md380tools/commit/d346969964b664c1155b6d76a49d7c94a98106c0

DL9SK, 2016-11-29:
Thank you very much. Enlarge to 19 and delete the spaces is enough for TG number.

DL9SK, 2016-11-29:
Please delete also the spaces before and after - > Then it will be good enough and the issue  could closed

sijskes, 2016-11-29:
fixed in D13.

DL9SK, 2016-11-30:
Everything now ok. Thank you very much for new code.

73, Ralf dl9SK

ISSUE CLOSED

closed 2016-11-30 by DL9SK