travisgoodspeed/md380tools #618: add "chierda D200"

ispkorte, 2016-11-25:
today i fix a D200 
its a MD380 clone 
difference to 380: R-SMA antenna and body-style 
[image]

travisgoodspeed, 2016-11-25:
Nice to know, but what is the software development issue here?

closed 2016-11-25 by travisgoodspeed

ispkorte, 2016-11-26:
sorry  Travis , for late respond. this was only a info for developers about this model .
Software is same md380/rt3 100% compatible

travisgoodspeed, 2016-11-26:
No worries.  Do we have a wiki page that lists known compatible models?  If not, perhaps we ought to start one.

roelandjansen, 2016-11-26:
It is on the main page and in the docs. Just add it.

sijskes, 2016-11-27:
https://github.com/travisgoodspeed/md380tools/wiki/MD380-Radios-and-variants