travisgoodspeed/md380tools #624: MD390/RT8 no Radiosettings in Utilities Menu 3

ispkorte, 2016-11-30:
[https://goo.gl/photos/iP6kxwffyxVaAEi26] here is 4 Pictures

first two is Menu from RT3
last two is Menu from RT8

on RT8 i have no option to change RX or TX settings (see attach picture)
[image]

anybody can confirm it ?

73 DD1GO

sijskes, 2016-11-30:
i see only 1 picture.

ispkorte, 2016-11-30:
check link on top in message

travisgoodspeed, 2016-11-30:
The channel editing feature is turned on and off by the codeplug in Tytera CPS.  Do your radios have different code plugs?

ispkorte, 2016-11-30:
is the same codeplug and some days before , menu 3 was available on RT8

ispkorte, 2016-11-30:
my codeplug -- > http://dd1go.de/RT3/Basis-CodePlug-RT3.rdt

ispkorte, 2016-12-07:
@travisgoodspeed  ; today i put the new (from now) firmware in my RT8 and now see Menu3 is coming back ; no codeplug change , only FW1e2c243

closed 2016-12-07 by ispkorte