travisgoodspeed/md380tools #626: screen MD380G up side down

on4amh, 2016-12-03:
i updated MD380G to experimental now my screen is upside down
[image]

TYT send me software "factory 2,04" now the problem upside down is solved but the letters are in mirror ( screen mirror )

[image]

what must i do?

aeickho, 2016-12-03:
Very interesting .. the difference of the used hardware components ... See #178 . 

Please let me know more about your radio.

- Shipped firmware version
- date of buying
- 

Your first picture is with not experimental fw (original font)
In the second picture there is the experimental fw running.

- alex, df8av

on4amh, 2016-12-03:
Hello Alex

I made a great fold, I did not look to the original firmware , when the
radio arrived yesterday everything was ok, I installed my code plug and it
worked, then I updated to experimental and than I get the up side
down
I installed back the s13.02 but no difference still upside down as you can
see on the picture

on4amh, 2016-12-03:
Firmware S13.02

on4amh, 2016-12-03:
down loaded date 3-12-2016

sijskes, 2016-12-03:
if screen type turns out to be non-detectable, then it would something for the menu. nice for party tricks and other mischiefs.

on4amh, 2016-12-04:
Where can i change settings for display see picture, 
[image]

aeickho, 2016-12-04:
the function address of  F_1075_display_init() on s13.020  
 is 0x08034086

you can see the different init code on  0x080340a4 and  0x080340e4 

cd annotations/s13.020
make (install radare2 bevor)

        ,=< 0x080340a4      03d5           bpl 0x80340ae               ;[4]
        |   0x080340a6      0820           movs r0, 8                  ; 8
        |   0x080340a8      fff7d0ff       bl 0x803404c                ;[3]
       ,==< 0x080340ac      02e0           b 0x80340b4                 ;[5]
       |`-> 0x080340ae      4820           movs r0, 0x48               ; 'H' ; 72
       |    0x080340b0      fff7ccff       bl 0x803404c                ;[3]
       `--> 0x080340b4      fe20           movs r0, 0xfe               ; 254
            0x080340b6      fff7c5ff       bl 0x8034044                ;[2]
            0x080340ba      ef20           movs r0, 0xef               ; 239

the type of display is configured by  the tuning parameter +0x1d &0x1 - see: #22 

I'll merge #623 now. And add printing of  the value of +0x1d

aeickho, 2016-12-04:
i see that 0x1d is not the offset from the TDFU_SPIFLASHSECURITYREGREAD

i'll try to understand the value of 0x2001df08 @ s13.020 -> with part of the SPIFLASHSECURITYREG is stored.

aeickho, 2016-12-04:
@on4amh please let me know the output from: 

./md380-tool readword 0x2001df25

on4amh, 2016-12-04:
Hello

Thats what i see

francis@francis-Aspire-5750:~/md380tools$ ./md380-tool readword 0x2001df25
WARNING: Unable to load user.bin.
2001df25
2001df25 00fffffc
2001df25 00 ff ff fc
2001df29 ff ff ff ff
2001df2d ff ff ff ff
2001df31 ff ff ff ff
16777212

73' ON4AMH
Francis

aeickho, 2016-12-04:
hy francis, 
  tnx.

can you please  repeat the test with "upside down" display ... what was the version of sw. 

So far we had only two different displays in the devices.

I have not heard of your observation yet.

possibilities:
- new 3. type of display
- wrong value in secure flash (device config)

question
- is there a newer version at tyt than the s013.020?
- how did the "up side down" is possible
-

Possible solutions:
- create a new init display function
- my
-- display_init_hook_1 () to 0x080340a8 and 0x080340b0 patches ...
-- display_init_hook_2 () to 0x803404c and 0x803404c
-- and then look, possibly new "additional parameter"

I do not understand what happened ...
Why only a radio ... ???

73, alex df8av

on4amh, 2016-12-04:
On the website is only the 13.020 I will ask it , when the radio arrived he was ok I loaded my code plug and everything worked normal , then I upgraded to experimental SW and the scree was flipped upside down but not revers, the radio is new, my idea is another version than 13.020
I did send a mail to TYT with pictures yesterday so I hope to get reply tomorrow 
Is will repeat the test tomorrow 
I'll keep you informed

aeickho, 2016-12-04:
you can see the difference between original and experimental simple by the different character fonts

the first picture in this issue is with original tyt sw
the second is with the experimental fw

on4amh, 2016-12-04:
new result i turned screen back

francis@francis-Aspire-5750:~/md380tools$ ./md380-tool readword 0x2001df25
WARNING: Unable to load user.bin.
2001df25
2001df25 00fffffd
2001df25 00 ff ff fd
2001df29 ff ff ff ff
2001df2d ff ff ff ff
2001df31 ff ff ff ff
16777213

> 
> you can see the difference between original and experimental simple by the different character fonts
> 
> the first picture in this issue is with original tyt sw
> the second is with the experimental fw

best 73

ON4AMH

aeickho, 2016-12-04:
With which fw?

on4amh, 2016-12-05:
Last Experimental firmware 

Vriendelijke groeten,

Francis Mahieu

aeickho, 2016-12-05:
@on4amh  .. can we close this tssue now?

on4amh, 2016-12-06:
You can close but the problem Still stays 

Vriendelijke groeten,

Francis Mahieu

on4amh, 2016-12-06:
dear Alex , i received today from TYT new software and the problem is solved with the original software
new SW : S13.32.bin
can you also make this work for the experimental

aeickho, 2016-12-06:
The display_init() of S13.032 @ 0x080340f4 has a third type of display included. I'll try to understand more about the new type of display yet.

aeickho, 2016-12-06:
I have included the display init function from s013.032  to s013.020-exp.  I'll double check the code tomorrow and then commit it.

aeickho, 2016-12-06:
https://github.com/aeickho/md380tools/commit/be8c6fbf52f967ad753d94577126827b7cef34b6

derAlbin, 2016-12-09:
Hey Ho! 

Here is a link to the Build (Yesterday) with aeickho's changes. Testet with a brand new MD380G shipped @23.11.16 from China. S/N 16070248*

Use with your own risk. 
https://drive.google.com/drive/folders/0B9pYFRz90RFKZzd2aU1yOHR6RnM?usp=sharing

on4amh, 2016-12-09:
Just installed and everything works ok

aeickho, 2016-12-09:
dear @on4amh , @derAlbin 
thanks for your rapport. I's now included on travis/master 

please recheck the master for correct working  on "new" radios.

thanks,
  alex

derAlbin, 2016-12-09:
Hey Ho,

The MD380(G) (Ver3?) up- side down problem is solved on Master. 
Thank you for your work Alex. 

73, DO2ASA

on4amh, 2016-12-10:
Works great many thanks 

73',
ON4AMH

Francis Mahieu

aeickho, 2016-12-12:
tnx for the report... 
 I'll close this issue now

closed 2016-12-12 by aeickho

fmalinski, 2016-12-12:
Please forgive me....I am very new to the TYT radios, new to DMR & Ham radios for that matter.  And it doesn't help any that I have a head injury from military service so I hope you will bear with me.  I have both the MD-390 VHF and UHF models.  I tried to update my firmware to a newer version I found on TYT888 webpage and ended up with an upside down screen.  While trying various other fw versions I kept the upside down screen (not backwards) and frequent reboot loop.  I have downloaded this fw posted by deAlbin to my VHF radio and it seems to work fine.   My questions are....should I be using this MD-380 fw on my MD-390 and exactly which one should I be using (I see 3 bin files).  Thank you so much and again please accept my apologies.

fmalinski, 2016-12-12:
Also, as a comment, can you correct the spelling of "Powering Down" in the version MD380G_Ver3_MD380Tools_081616.bin

sijskes, 2016-12-13:
@fmalinski this is not the place to get support for your problems. 
You should try https://groups.google.com/group/md380tools